openbmc/openbmc #2504: Witherspoon: PNOR code update failed

yadlapati, 2017-10-23:
Seeing error while attempting PNOR upgrade
 curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/8418a6ef": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Failed", 
      "ExtendedVersion": "op-build-v1.7-1510-g7ac9dc9,buildroot-2017.08-8-g5e23247,skiboot-v5.9-rc4,hostboot-820a494,linux-4.13.7-openpower1-p5a9ada1,petitboot-v1.6.1-p14344ad,machine-xml-eee619a,occ-6404302,hostboot-binaries-c5ff921,capp-ucode-p9-dd2-v2,sbe-ec9a99e", 
      "Path": "/tmp/images/8418a6ef", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active", 
      "Version": "IBM-witherspoon-ibm-OP9_v1.19_1.105", 
      "associations": [
        [
          "inventory", 
          "activation", 
          "/xyz/openbmc_project/inventory/system/chassis"
        ]
      ]

BMC: w50.aus.stglabs.ibm.com

FFDC is /gsa/ausgsa/home/u/m/umay/web/public/ffdc_2017-10-18_13-07-14.tar

saqibkh, 2017-10-23:
Oct 18 13:03:23 witherspoon obmc-flash-bios[1600]: mount: wrong fs type, bad option, bad superblock on ubi8:pnor-prsv,
Oct 18 13:03:23 witherspoon obmc-flash-bios[1600]:        missing codepage or helper program, or other error
Oct 18 13:03:23 witherspoon obmc-flash-bios[1600]:        In some cases useful info is found in syslog - try
Oct 18 13:03:23 witherspoon obmc-flash-bios[1600]:        dmesg | tail or so.
Oct 18 13:03:23 witherspoon obmc-flash-bios[1600]: /usr/sbin/obmc-flash-bios: error 32
Oct 18 13:03:23 witherspoon systemd[1]: obmc-flash-bios-ubimount@8418a6ef.service: Main process exited, code=exited, status=32/n/a
Oct 18 13:03:23 witherspoon systemd[1]: Failed to start Mount UBIFS volumes pnor-ro-8418a6ef, pnor-rw-8418a6ef and pnor-prsv.
Oct 18 13:03:23 witherspoon kernel: UBIFS error (ubi8:2 pid 1609): ubifs_read_node: bad node type (255 but expected 6)
Oct 18 13:03:23 witherspoon kernel: UBIFS error (ubi8:2 pid 1609): ubifs_read_node: bad node at LEB 0:0, LEB mapping status 0
Oct 18 13:03:23 witherspoon kernel: Not a node, first 24 bytes:
Oct 18 13:03:23 witherspoon kernel: 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                          ........................
Oct 18 13:03:23 witherspoon kernel: CPU: 0 PID: 1609 Comm: mount Not tainted 4.10.17-703c3c181790b20c48c4d02b9fed901379671fb7 #1
Oct 18 13:03:23 witherspoon kernel: Hardware name: ASpeed SoC
Oct 18 13:03:23 witherspoon kernel: [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
Oct 18 13:03:23 witherspoon kernel: [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
Oct 18 13:03:23 witherspoon kernel: [<801dbf2c>] (dump_stack) from [<80170e30>] (ubifs_read_node+0x240/0x2f8)
Oct 18 13:03:23 witherspoon kernel: [<80170e30>] (ubifs_read_node) from [<8016d4ec>] (ubifs_read_sb_node+0x64/0x8c)
Oct 18 13:03:23 witherspoon kernel: [<8016d4ec>] (ubifs_read_sb_node) from [<8016dc3c>] (ubifs_read_superblock+0x6d8/0x10e0)
Oct 18 13:03:23 witherspoon kernel: [<8016dc3c>] (ubifs_read_superblock) from [<8016be14>] (ubifs_mount+0x96c/0x1d98)
Oct 18 13:03:23 witherspoon kernel: [<8016be14>] (ubifs_mount) from [<800e658c>] (mount_fs+0x24/0xb8)
Oct 18 13:03:23 witherspoon kernel: [<800e658c>] (mount_fs) from [<80102040>] (vfs_kern_mount+0x58/0x100)
Oct 18 13:03:23 witherspoon kernel: [<80102040>] (vfs_kern_mount) from [<801038a0>] (do_mount+0x160/0xb2c)
Oct 18 13:03:23 witherspoon kernel: [<801038a0>] (do_mount) from [<8010460c>] (SyS_mount+0x9c/0xc4)
Oct 18 13:03:23 witherspoon kernel: [<8010460c>] (SyS_mount) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
Oct 18 13:03:23 witherspoon systemd[1]: obmc-flash-bios-ubimount@8418a6ef.service: Unit entered failed state.
Oct 18 13:03:23 witherspoon kernel[967]: [ 1101.820000] UBIFS error (ubi8:2 pid 1609): ubifs_read_node: bad node type (255 but expected 6)
Oct 18 13:03:23 witherspoon systemd[1]: obmc-flash-bios-ubimount@8418a6ef.service: Triggering OnFailure= dependencies.
Oct 18 13:03:23 witherspoon kernel[967]: [ 1101.820000] UBIFS error (ubi8:2 pid 1609): ubifs_read_node: bad node at LEB 0:0, LEB mapping status 0
Oct 18 13:03:23 witherspoon systemd[1]: obmc-flash-bios-ubimount@8418a6ef.service: Failed with result 'exit-code'.
Oct 18 13:03:23 witherspoon kernel[967]: [ 1101.820000] Not a node, first 24 bytes:
Oct 18 13:03:23 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-8418a6ef...
Oct 18 13:03:23 witherspoon kernel[967]: [ 1101.820000] 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                          ........................
Oct 18 13:03:23 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-8418a6ef...

saqibkh, 2017-10-23:
manually deleting the pnor-prsv seems to have resolved this issue. However the code update script should have been able to handle this scenario.

amboar, 2017-10-23:
@yadlapati This is a public issue tracker; can you please make sure to attach any relevant artefacts and provide the version information such that someone can recreate your issue without needing access to internal IBM systems to analyse the problem. Specifically,

> BMC: w50.aus.stglabs.ibm.com
>
> FFDC is /gsa/ausgsa/home/u/m/umay/web/public/ffdc_2017-10-18_13-07-14.tar

is unhelpful to people outside of IBM that might be interested in your issue.

anoo1, 2018-08-15:
A lot of code update fixes have gone in since last year, please retry with the latest mater.

closed 2018-08-15 by anoo1