openbmc/openbmc #2505: Time Manager symbolic link mismatch

gkeishin, 2017-10-23:
```
Oct 12 12:02:05 witherspoon dbus[960]: [system] Successfully activated service 'org.freedesktop.timedate1'
Oct 12 12:02:05 witherspoon systemd[1]: Started Time & Date Service.
Oct 12 12:02:05 witherspoon systemd-timedated[1324]: /etc/localtime should be a symbolic link to a time zone data file in /usr/share/zoneinfo/.
Oct 12 12:02:05 witherspoon systemd[1]: Started Phosphor Time Manager daemon.
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-174"
VERSION_ID="v1.99.10-174-gc9f3377"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-174"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-da78428ac543d6d9ac2493dfa77364eb300a47cb #1 Fri Oct 20 20:00:41 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-b0507b6c/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.68
        op-build-v1.7-1473-g6f195a1
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9-rc3
        hostboot-b01e6bb
        linux-4.13.5-openpower1-pc511380
        petitboot-v1.6.0-p3f89369
        machine-xml-389c56f
        occ-ba4e81e
        hostboot-binaries-da1227e
        capp-ucode-p9-dd2-v2
        sbe-374add4
root@witherspoon:~#
```

- [x] Journald log 
[RR_BMC_error.txt](https://github.com/openbmc/openbmc/files/1408578/RR_BMC_error.txt)

mine260309, 2017-10-25:
It is not related to time manager, but missing the time zone package.

gkeishin, 2017-10-25:
So what is the next action it needed to be done for this ? @mine260309

mine260309, 2017-10-25:
Better to let @geissonator see which one should investigate this issue, or decide if we can just ignore this warning message.

timlee66, 2018-09-13:
@mine260309, could you kindly help to point out where is the time zone package? 
Due to I find that /etc/locatime this symbolink not exist in openbmc and also /usr/share/zoneinfo/
Thus, when I use "timedatectl list-timezones" then only got "UTC" result. However, I cannot use "timedatectl set-timezones" to change TZ for openbmc now. Is there any I miss install to openbmc?

Thanks
Tim Lee

mine260309, 2018-09-13:
@timlee66 Most likely it is missing tzdata package.
Could you try to add `tzdata`, and `tzdata-<area>` in your build and see if it works?
Refer to `poky/meta/recipes-extended/tzdata/tzdata_2018d.bb` for package details.

mdmillerii, 2018-09-13:
I think many systems do not support time zones and only support the default `UTC` / `GMT` as the bmc system timezone.

timlee66, 2018-09-14:
Hello @mine260309, thanks your response so quickly and it work when I build tzdata package. timedatectl list-timezones and timedatectl set-timezone both can work correctly. Thank your help!!

Best regards,
Tim

gkeishin, 2019-04-15:
@mine260309 anything on this

mine260309, 2019-04-16:
I do not think we will fix this, because tzdata cost a lot of flash space.

closed 2019-11-15 by stale[bot]