openbmc/openbmc #2506: DHCP get wrong domain.

KenChenIEC, 2017-10-24:
My domain should be 10.6.45.xx, but the lastest openbmc code base will get 169.254.163.43 as below.
[image]

I changed the commit ID in meta-phosphor/classes/phosphr-networkd-rev.bbclass to c9645fe5c874930445bca6ba51ac9daee916c99b, which is older version.
The IP is normally.

ratagupt, 2017-10-24:
I need the out put of "ipaddr show dev eth0" instead of ifconfig.
Ifconfig is deprecated now.

amboar, 2017-10-24:
@KenChenIEC For what it's worth 169.254.0.0/16 is the IPv4 link-local network block. Also, multiple IPs can be bound to a single interface, so it looks like you just need your usual network configuration applied.

@KenChenIEC Also, please try to refrain from posting images where possible, ideally the text from the console should be put into the issue. Given this is a networking problem I appreciate that might be harder, but given you're on the console you're probably viewing that on a different machine anyway.

KenChenIEC, 2017-10-25:
@ratagupt Thanks you updated. it's work on "ip addr show dev eth0"

> root@2400-rep:~# ip addr show dev eth0
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
>     link/ether d6:78:56:a6:ec:66 brd ff:ff:ff:ff:ff:ff
>     inet 169.254.197.174/16 brd 169.254.255.255 scope link eth0
>        valid_lft forever preferred_lft forever
>     inet 10.6.45.80/24 brd 10.6.45.255 scope global dynamic eth0
>        valid_lft 259173sec preferred_lft 259173sec
>     inet6 fe80::d478:56ff:fea6:ec66/64 scope link tentative flags 08
>        valid_lft forever preferred_lft forever

@amboar Got it, I will post the text in future.

ratagupt, 2017-10-25:
@KenChenIEC Can we close this issue now?

KenChenIEC, 2017-10-26:
@ratagupt Sure, Thanks.

closed 2017-10-26 by KenChenIEC