openbmc/openbmc #2507: IPMID logging footprint

gkeishin, 2017-10-24:
Can we reduce or silence some of the IPMID Footprint on the journald?

May be start filter some out starting from this logs for example.

```
root@witherspoon:~# journalctl --no-pager | grep -i ipmid
Oct 12 12:02:07 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:02:07 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:02:07 witherspoon ipmid[1290]: Asserting SMS Attention
Oct 12 12:02:08 witherspoon ipmid[1290]: SMS Attention asserted
Oct 12 12:02:12 witherspoon ipmid[1290]: Host control timeout hit!
Oct 12 12:02:12 witherspoon ipmid[1290]: Timer expired
Oct 12 12:03:08 witherspoon ipmid[1290]: Control Host work queue is empty!
Oct 12 12:03:10 witherspoon ipmid[1290]: Incomplete Fru file.. Size:[56]
Oct 12 12:03:10 witherspoon ipmid[1290]: Populating FRU areas failed for:[1]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering handler:[/usr/lib/host-ipmid/libsysintfcmds.so.0]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x35]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x2E]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x31]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering handler:[/usr/lib/host-ipmid/libstrgfnhandler.so.1]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x12]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering handler:[/usr/lib/host-ipmid/liboemhandler.so.0]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x32], Cmd:[0xF0]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x32], Cmd:[0x10]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering handler:[/usr/lib/host-ipmid/libapphandler.so.0]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x36]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0xFF]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x22]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x24]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x1]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x4]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x8]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x6]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x40]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x41]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x6], Cmd:[0x42]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0xFF]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x2F]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x30]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x2D]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x22]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x20]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x4], Cmd:[0x21]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0xFF]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x40]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x48]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x49]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x42]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x43]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x46]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x44]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x47]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x10]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xA], Cmd:[0x11]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0xFF]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0x0]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0x9]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0x1]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0x2]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x0], Cmd:[0x8]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x3]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x4]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x5]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x6]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x8]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0x9]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0x2C], Cmd:[0xA]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xC], Cmd:[0xFF]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xC], Cmd:[0x1]
Oct 12 12:03:17 witherspoon ipmid[1290]: Registering NetFn:[0xC], Cmd:[0x2]
[... 192 lines cut ...]
Oct 12 12:04:34 witherspoon ipmid[1290]: getHostNetworkData failed for get_sys_boot_options.
Oct 12 12:04:34 witherspoon ipmid[1290]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Oct 12 12:04:36 witherspoon ipmid[1290]: getHostNetworkData failed for get_sys_boot_options.
Oct 12 12:04:36 witherspoon ipmid[1290]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Oct 12 12:04:38 witherspoon ipmid[1290]: getHostNetworkData failed for get_sys_boot_options.
Oct 12 12:04:38 witherspoon ipmid[1290]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Oct 12 12:04:38 witherspoon ipmid[1290]: No Registered handlers for NetFn:[0x3A],Cmd:[0xFF]
Oct 12 12:04:38 witherspoon ipmid[1290]: ERROR:[0xC1] handling NetFn:[0x3A], Cmd:[0x1A]
Oct 12 12:04:55 witherspoon ipmid[1290]: getHostNetworkData failed for get_sys_boot_options.
Oct 12 12:04:55 witherspoon ipmid[1290]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Oct 12 12:05:08 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:05:08 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:05:08 witherspoon ipmid[1290]: Asserting SMS Attention
Oct 12 12:05:17 witherspoon ipmid[1290]: SMS Attention asserted
Oct 12 12:05:18 witherspoon ipmid[1290]: Failed to call the method: Input/output error
Oct 12 12:05:18 witherspoon ipmid[1290]: Dest: :1.2, Path: /org/openbmc/HostIpmi/1
Oct 12 12:05:19 witherspoon ipmid[1290]: Soft off is running, so let shutdown target stop the host
Oct 12 12:05:42 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:05:42 witherspoon ipmid[1290]: Pushing cmd on to queue
Oct 12 12:05:42 witherspoon ipmid[1290]: Asserting SMS Attention
Oct 12 12:05:43 witherspoon ipmid[1290]: SMS Attention asserted
Oct 12 12:05:46 witherspoon ipmid[1290]: Host control timeout hit!
Oct 12 12:05:46 witherspoon ipmid[1290]: Timer expired
root@witherspoon:~#
```

aditya0124, 2017-10-26:
Which messages should be silenced.
Are those trace messages identified ?

dkodihal, 2017-10-26:
> Are those trace messages identified ?

Check the snippet attached to this issue and get feedback from @tomjoseph83.

gkeishin, 2017-10-26:
As a start, run the IPL, reboot Host and reset reload BMC when host running, grab those journald snapshots and review what can be removed and what can be improved and what can be retained.

aditya0124, 2017-10-26:
Sure

gkeishin, 2017-11-03:
```
Oct 26 19:54:19 witherspoon ipmid[1365]: Successfully verified area checksum. offset:[8]
Oct 26 19:54:19 witherspoon ipmid[1365]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru14]
Oct 26 19:54:19 witherspoon ipmid[1365]: fru data size:[56], area offset:[8], area_size:[24Incomplete Fru file.. Size:[56]
Oct 26 19:54:19 witherspoon ipmid[1365]: Populating FRU areas failed for:[26]
Oct 26 19:54:19 witherspoon ipmid[1365]: Incomplete Fru file.. Size:[112]
Oct 26 19:54:19 witherspoon ipmid[1365]: Populating FRU areas failed for:[26]
Oct 26 19:54:19 witherspoon ipmid[1365]: Incomplete Fru file.. Size:[168]
Oct 26 19:54:19 witherspoon ipmid[1365]: Populating FRU areas failed for:[26]
Oct 26 19:54:19 witherspoon ipmid[1365]: Incomplete Fru file.. Size:[224]
Oct 26 19:54:19 witherspoon ipmid[1365]: Populating FRU areas failed for:[26]
Oct 26 19:54:19 witherspoon ipmid[1365]: 8]
Oct 26 19:54:19 witherspoon ipmid[1365]: fru data size:[112], area offset:[8], area_size:[248]
```

some more inputs of incomplete traces for IPMID during IPL

aditya0124, 2017-11-07:
I did a fresh reboot of the W15 .. Didn't copied any of my local binaries .. and got the following footprint for the ipmid logs ..

root@witherspoon:~# journalctl --no-pager | grep -i ipmid
Nov 07 06:50:41 witherspoon ipmid[1345]: Pushing cmd on to queue
Nov 07 06:50:41 witherspoon ipmid[1345]: Pushing cmd on to queue
Nov 07 06:50:41 witherspoon ipmid[1345]: Asserting SMS Attention
Nov 07 06:50:41 witherspoon ipmid[1345]: SMS Attention asserted
Nov 07 06:50:45 witherspoon ipmid[1345]: Host control timeout hit!
Nov 07 06:50:45 witherspoon ipmid[1345]: Timer expired
root@witherspoon:~# 

Am I missing something ... Do I need to run/start any additional process to get the required logs

gkeishin, 2017-11-07:
Probably the RTC clock thing messing with your journald.. 

`hwclock --utc --systohc`
`systemctl stop systemd-journald.service && rm -rf /var/log/journal && systemctl start systemd-journald.service`
and IPL again..

rfrandse, 2017-11-09:
<https://gerrit.openbmc-project.xyz/7850> remove 'WATCHDOG RESET' message
Resolves  #2507 IPMID logging footprint

amboar, 2017-11-13:
@aditya0124 Please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

mine260309, 2018-01-03:
Does anyone know why there are below logs, are they real errors or can be ignored?

```
Oct 12 12:03:17 witherspoon ipmid[1290]: LOOKATME: Sensor should not be reported:  Type 0x03, Offset 0x01
```
and
```
Oct 12 12:03:19 witherspoon ipmid[1290]: Incomplete Fru file.. Size:[56]
Oct 12 12:03:19 witherspoon ipmid[1290]: Populating FRU areas failed for:[2]
Oct 12 12:03:19 witherspoon ipmid[1290]: Incomplete Fru file.. Size:[56]
Oct 12 12:03:19 witherspoon ipmid[1290]: Populating FRU areas failed for:[3]
Oct 12 12:03:19 witherspoon ipmid[1290]: Incomplete Fru file.. Size:[112]
```
and
```
Oct 12 12:04:31 witherspoon ipmid[1290]: getHostNetworkData failed for get_sys_boot_options.
Oct 12 12:04:31 witherspoon ipmid[1290]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
```
and
```
Oct 12 12:05:18 witherspoon ipmid[1290]: Failed to call the method: Input/output error
Oct 12 12:05:18 witherspoon ipmid[1290]: Dest: :1.2, Path: /org/openbmc/HostIpmi/1
```

rfrandse, 2018-01-03:
<https://gerrit.openbmc-project.xyz/7850> remove 'WATCHDOG RESET' message
Partially Resolves  #2507 IPMID logging footprint

rfrandse, 2018-03-07:
<https://gerrit.openbmc-project.xyz/7850> Reducing IPMI logging footprint
Resolves  #2507 IPMID logging footprint

dkodihal, 2018-04-16:
https://gerrit.openbmc-project.xyz/#/c/7850/ is still pending.

closed 2018-04-18 by geissonator