openbmc/openbmc #2511: Bunch of File system unmount failure on BMC reset

gkeishin, 2017-10-24:
Boot to Standby, reset BMC. This is seen in all Master build for sometime. Time to look at this one

```
[  OK  ] Stopped target System Initialization.
         Stopping Network Time Synchronization...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Load/Save Random Seed...
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting Temporary Directory...
         Unmounting /media/pnor-prsv...
         Unmounting /usr/local/share/pnor...
         Unmounting /media/pnor-rw-b0507b6c...
         Unmounting /etc/machine-id...
[ 2091.010000] UBIFS (ubi8:1): un-mount UBI device 8
[ 2091.010000] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
[ 2091.010000] UBIFS (ubi8:3): un-mount UBI device 8
[ 2091.070000] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" stops
[ 2091.070000] UBIFS (ubi8:2): un-mount UBI device 8
[ 2091.140000] UBIFS (ubi8:2): background thread "ubifs_bgt8_2" stops
         Unmounting /media/rofs-956a53bf...
         Unmounting /var/volatile...
         Unmounting /media/pnor-ro-b0507b6c...
         Unmounting /home...
[  OK  ] Stopped Flush Journal to Persistent Storage.
[  OK  ] Unmounted Temporary Directory.
[  OK  ] Unmounted /media/pnor-prsv.
[  OK  ] Unmounted /usr/local/share/pnor.
[  OK  ] Unmounted /media/pnor-rw-b0507b6c.
[FAILED] Failed unmounting /etc/machine-id.
[  OK  ] Unmounted /media/rofs-956a53bf.
[FAILED] Failed unmounting /home.
[FAILED] Failed unmounting /var/volatile.
[FAILED] Failed unmounting /media/pnor-ro-b0507b6c.
         Unmounting /var...
         Unmounting /etc...
         Unmounting /usr/local...
         Unmounting /media...
[  OK  ] Stopped target Swap.
[  OK  ] Unmounted /var.
[  OK  ] Unmounted /media.
[  OK  ] Unmounted /usr/local.
[  OK  ] Unmounted /etc.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[ 2093.190000] watchdog: watchdog0: watchdog did not stop!
[ 2093.190000] systemd-shutdow: 50 output lines suppressed due to ratelimiting
[ 2093.310000] systemd-journald[591]: Failed to send WATCHDOG=1 notification message: Connection refused
[ 2093.310000] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[ 2093.440000] systemd-journald[591]: Received SIGTERM from PID 1 (systemd-shutdow).
[ 2093.440000] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[ 2093.440000] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[ 2093.440000] systemd-shutdown[1]: Unmounting file systems.
[ 2093.440000] systemd-shutdown[1]: Remounting '/home' read-only with options ''.
[ 2093.440000] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[ 2093.580000] systemd-shutdown[1]: Unmounting /home.
[ 2093.580000] UBIFS (ubi0:2): un-mount UBI device 0
[ 2093.580000] systemd-shutdown[1]: Remounting '/etc/machine-id' read-only with options 'mode=755'.
[ 2093.580000] systemd-shutdown[1]: Unmounting /etc/machine-id.
[ 2093.580000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[ 2093.580000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[ 2093.730000] reboot: Restarting system

U-Boot 2016.07 (Aug 29 2017 - 21:30:45 +0000)

```

eddiejames, 2017-10-24:
I think this is quite common, it doesn't seem to affect anything. I'm not sure why the unmounts fail; is it worth investigating at the moment?

gkeishin, 2017-10-24:
@eddiejames Not a priority one for now. For reference logging, the failure observed for now.

gkeishin, 2017-10-31:
```
[FAILED] Failed to start Prepare for overlay fs.
See 'systemctl status prepare-overlay.service' for details.
[DEPEND] Dependency failed for /etc.
[DEPEND] Dependency failed for Local File Systems.
[DEPEND] Dependency failed for Mount machine-id.
[DEPEND] Dependency failed for /home.
[DEPEND] Dependency failed for Emergency Shell.
[DEPEND] Dependency failed for Emergency Mode.
[  OK  ] Closed Syslog Socket.
```
Now its really bricked the system on few reboots. Attaching the last current serial console log

- [x] Journald 
[SERIAL_CONSOLE_HANG.txt](https://github.com/openbmc/openbmc/files/1430950/SERIAL_CONSOLE_HANG.txt)

gkeishin, 2017-11-10:
```
Nov 09 22:05:26 witherspoon systemd[1]: Unmounting /media/pnor-prsv...
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:2): un-mount UBI device 8
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:2): background thread "ubifs_bgt8_2" stops
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /home...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /media/rofs-63521bb7...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /media/pnor-ro-18f3a974...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /media/pnor-rw-18f3a974...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /usr/local/share/pnor...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /media/rofs-cd97a4c9...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /var/volatile...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting Temporary Directory...
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:3): un-mount UBI device 8
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:3): background thread "ubifs_bgt8_3" stops
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /media/pnor-prsv.
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:1): un-mount UBI device 8
Nov 09 22:05:27 witherspoon kernel: UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
Nov 09 22:05:27 witherspoon systemd[1]: home.mount: Mount process exited, code=exited status=1
Nov 09 22:05:27 witherspoon systemd[1]: Failed unmounting /home.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /media/rofs-63521bb7.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /media/pnor-ro-18f3a974.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /usr/local/share/pnor.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /media/rofs-cd97a4c9.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted Temporary Directory.
Nov 09 22:05:27 witherspoon systemd[1]: Stopped target Swap.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounting /usr/local...
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /media/pnor-rw-18f3a974.
Nov 09 22:05:27 witherspoon systemd[1]: var-volatile.mount: Mount process exited, code=exited status=1
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /var/volatile.
Nov 09 22:05:27 witherspoon systemd[1]: var-volatile.mount: Unit entered failed state.
Nov 09 22:05:27 witherspoon systemd[1]: Unmounted /usr/local.
Nov 09 22:05:28 witherspoon systemd[1]: Unmounting /var...
Nov 09 22:05:28 witherspoon systemd[1]: Unmounting /media...
Nov 09 22:05:28 witherspoon systemd[1]: var.mount: Mount process exited, code=exited status=1
Nov 09 22:05:28 witherspoon systemd[1]: Failed unmounting /var.
Nov 09 22:05:28 witherspoon systemd[1]: Unmounted /media.
Nov 09 22:05:28 witherspoon systemd[1]: Reached target Unmount All Filesystems.
Nov 09 22:05:28 witherspoon systemd[1]: Stopped target Local File Systems (Pre).
Nov 09 22:05:28 witherspoon systemd[1]: Stopped Create Static Device Nodes in /dev.
Nov 09 22:05:28 witherspoon systemd[1]: Stopped Create System Users.
Nov 09 22:05:28 witherspoon systemd[1]: Stopped Remount Root and Kernel File Systems.
Nov 09 22:05:28 witherspoon systemd[1]: Reached target Shutdown.
Nov 09 22:05:28 witherspoon systemd[1]: Reached target Final Step.
Nov 09 22:05:28 witherspoon systemd[1]: Starting Reboot...
Nov 09 22:05:28 witherspoon systemd[1]: Shutting down.
Nov 09 22:05:29 witherspoon systemd[1]: Hardware watchdog 'aspeed_wdt', version 0
Nov 09 22:05:29 witherspoon systemd[1]: Set hardware watchdog to 10min.
Nov 09 22:05:29 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Nov 09 22:05:29 witherspoon kernel: systemd-shutdow: 68 output lines suppressed due to ratelimiting
Nov 09 22:05:30 witherspoon systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Nov 09 22:05:30 witherspoon systemd-journald[939]: Journal stopped
```

geissonator, 2018-03-30:
Still seeing this with latest 4.13 kernel:
```
[  OK  ] Stopped Bind mount volatile /srv.
[FAILED] Failed unmounting /media/pnor-prsv.
[FAILED] Failed unmounting /usr/local/share/pnor.
[FAILED] Failed unmounting /media/rofs-aa042154.
[FAILED] Failed unmounting /media/pnor-ro-8764baa3.
[FAILED] Failed unmounting /media/pnor-ro-4817795c.
[FAILED] Failed unmounting /media/rofs-331ef330.
[FAILED] Failed unmounting /home.
[FAILED] Failed unmounting /media/pnor-rw-8764baa3.
[  OK  ] Unmounted /srv.
[FAILED] Failed unmounting /media/pnor-rw-4817795c.
[FAILED] Failed unmounting Temporary Directory (/tmp).
[  OK  ] Stopped target Swap.
         Unmounting /usr/local...
         Unmounting /media...
         Unmounting /var/volatile...
[FAILED] Failed unmounting /usr/local.
[FAILED] Failed unmounting /media.
[FAILED] Failed unmounting /var/volatile.
         Unmounting /var...
[FAILED] Failed unmounting /var.
```

gkeishin, 2018-04-04:
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-e7ae4c9b5df12838e66a25df93df02e47bdb7d11 #1 Fri Mar 23 11:41:04 CDT 2018 armv6l GNU/Linux
root@witherspoon:~#

[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /var/volatile...
         Unmounting /media/rofs-78d09908...
         Unmounting /media/pnor-rw-4f901d3d...
[70551.190000] UBIFS (ubi8:1): un-mount UBI device 8
[70551.190000] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops
         Unmounting /media/pnor-prsv...
[70551.320000] UBIFS (ubi8:2): un-mount UBI device 8
[70551.320000] UBIFS (ubi8:2): background thread "ubifs_bgt8_2" stops
         Unmounting /media/pnor-ro-4f901d3d...
         Unmounting /home...
         Unmounting Temporary Directory...
[  OK  ] Unmounted /var/volatile.
[  OK  ] Unmounted /media/rofs-78d09908.
[  OK  ] Unmounted /media/pnor-rw-4f901d3d.
[  OK  ] Unmounted /media/pnor-prsv.
[  OK  ] Unmounted /media/pnor-ro-4f901d3d.
[  OK  ] Unmounted /home.
[  OK  ] Unmounted Temporary Directory.
[  OK  ] Stopped target Swap.
         Unmounting /media...
         Unmounting /var...
[  OK  ] Unmounted /media.
[FAILED] Failed unmounting /var.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[70552.820000] watchdog: watchdog0: watchdog did not stop!
[70552.820000] systemd-shutdow: 51 output lines suppressed due to ratelimiting
[70552.920000] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[70553.150000] systemd-journald[928]: Received SIGTERM from PID 1 (systemd-shutdow).
[70553.280000] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[70553.280000] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[70553.570000] systemd-shutdown[1]: Unmounting file systems.
[70553.570000] systemd-shutdown[1]: Remounting '/var' read-only with options ''.
[70553.570000] UBIFS (ubi0:0): background thread "ubifs_bgt0_0" stops
[70553.570000] systemd-shutdown[1]: Unmounting /var.
[70553.890000] UBIFS (ubi0:0): un-mount UBI device 0
[70553.890000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[70553.890000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[70553.890000] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[70553.890000] systemd-shutdown[1]: All filesystems unmounted.
[70553.890000] reboot: Restarting system

U-Boot 2016.07 (Mar 29 2018 - 12:06:59 -0500)

@amboar

eddiejames, 2018-04-04:
I have a fix up for this. Needed an upstream busybox patch.

This will NOT fix the /var failure, which is due to the journal still logging during shutdown. It isn't possible to fix this without introducing side-effects; see https://github.com/systemd/systemd/issues/867. The final shutdown procedure mounts read-only and then unmounts anyway, after the journal has stopped, so the issue is purely cosmetic.

In fact, all of these errors were purely cosmetic, as systemd will always clean up during final shutdown...

rfrandse, 2018-04-04:
<https://gerrit.openbmc-project.xyz/9934> busybox: patch "umount: ignore -c"
Resolves  #2511 Bunch of File system unmount failure on BMC reset

closed 2018-04-10 by geissonator

gkeishin, 2018-04-10:
Nice.. let me verify this on the master public build.

gkeishin, 2018-04-10:
Super clean as compare to previous..  see's only /var unmount failure but can't complain much.. Thanks, @eddiejames 

```
~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-396"
VERSION_ID="v2.1-396-g4e9d7e6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-396"
BUILD_ID="v2.1"

~# uname -a
Linux wsbmc011 4.13.16-158c8e15d577cebdb047f5d553e12054b15b1b84 #1 Thu Apr 5 17:22:57 UTC 2018 armv6l GNU/Linux
```
From Serial
```

         Unmounting /media/rofs-3289ac99...
         Unmounting /home...
         Unmounting /srv...
         Unmounting Temporary Directory (/tmp)...
[  OK  ] Unmounted /media/pnor-ro-4f901d3d.
[  OK  ] Unmounted /media/pnor-prsv.
[  OK  ] Unmounted /media/rofs-c93781be.
[  OK  ] Unmounted /media/pnor-rw-4f901d3d.
[  OK  ] Stopped Bind mount volatile /srv.
[  OK  ] Unmounted /media/rofs-3289ac99.
[  OK  ] Unmounted /home.
[  OK  ] Unmounted /srv.
[  OK  ] Unmounted Temporary Directory (/tmp).
[  OK  ] Stopped target Swap.
         Unmounting /var/volatile...
         Unmounting /media...
[  OK  ] Unmounted /var/volatile.
         Unmounting /var...
[  OK  ] Unmounted /media.
[FAILED] Failed unmounting /var.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[  284.444325] watchdog: watchdog0: watchdog did not stop!
[  284.477459] systemd-shutdow: 53 output lines suppressed due to ratelimiting
[  284.899438] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  284.930130] systemd-journald[660]: Received SIGTERM from PID 1 (systemd-shutdow).
[  285.194082] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  285.219720] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[  285.232471] systemd-shutdown[1]: Unmounting file systems.
[  285.239107] systemd-shutdown[1]: Remounting '/var' read-only with options 'ubi=0,vol=2'.
[  285.397951] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[  285.535180] systemd-shutdown[1]: Unmounting /var.
[  285.545697] UBIFS (ubi0:2): un-mount UBI device 0
[  285.552874] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[  285.569574] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[  285.577165] systemd-shutdown[1]: Remounting '/' read-only with options ''.
[  285.584808] systemd-shutdown[1]: All filesystems unmounted.
[  285.631618] reboot: Restarting system

U-Boot 2016.07 (Apr 10 2018 - 04:21:23 +0000)

       Watchdog enabled
DRAM:  496 MiB
```

anoo1, 2018-06-26:
> Super clean as compare to previous.. see's only /var unmount failure but can't complain much..

Per Eddie, /var will always have an error, it is a known limitation of upstream systemd.