openbmc/openbmc #2516: RAS:OP910: Flood of PowerSupplyUnderVoltageFault SELs logged in BMC on pulling a PSU

cdeadmin, 2017-10-25:


bjwyman, 2017-10-25:
When a Witherspoon system has the AC power cable removed from the power supply, the expectation is for a single SEL to be logged, and some kind of physical indicator (LED on) of a problem.

The current implementation of the power fault analysis application (witherspoon-psu-monitor) is logging multiple errors. The fault for the VIN_UV_FAULT is getting logged, but then the power supply is not on (UNIT_IS_OFF) and the PGOOD is not active (POWER_GOOD Negated), so another fault gets logged for that. Having these multiple faults just adds to confusion, the single fault for the VIN_UV_FAULT should suffice in this case.

The second fault (power supply should be on) does have a call out, but if the power supply is removed (not present) then re-inserted/replaced (present), that fault is not cleared. The fault for the power supply not being on should be cleared if the power supply presence changes. In addition to that, the VIN_UV_FAULT case will need to have a call out in order for the enclosure light/LED to turn on as a physical indication of the problem with missing power.

bjwyman, 2017-10-25:
# Task List
- [ ] Clear powerOnFault on presence update
- [ ] Add call out to input/UV fault.
- [ ] Change to not log powered on faults if input/UV fault.

bjwyman, 2017-10-25:
# **STATUS**
**10/25**
- Today: Discussed multiple faults issue during a meeting with Jordan and Mike on power supply faults. Agreed that multiple faults, especially the power supply off after AC removed, adds to confusion. Created list of three things to change. Pushed up commit to clear the powerOnFault when power supply goes from missing to present. It was missing from the list.
- Tomorrow: Test ShouldBeOn fault clearing if power supply removed and replaced. Continue on task list.
- Blockers: none

rfrandse, 2017-11-02:
<https://gerrit.openbmc-project.xyz/7742> Fix up the inventory path
Resolves:  #2516 RAS:OP910: Flood of PowerSupplyUnderVoltageFault SELs logged in BMC on pulling a PSU

bjwyman, 2017-11-09:
https://gerrit.openbmc-project.xyz/7787 actually resolves this.

closed 2017-11-09 by geissonator