openbmc/openbmc #2517: phosphor-inventory JSON Parsing failed - provided NVP (cereal_class_version) not found

gkeishin, 2017-10-25:
Update to latest Master https://openpower.xyz/job/openbmc-build/1065/ and on boot to standby, we started seeing tons of this logged in journald.

Not something we would want to see bunch logging

```
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
Oct 12 12:03:08 witherspoon phosphor-inventory[1029]: JSON Parsing failed - provided NVP (cereal_class_version) not found
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-200"
VERSION_ID="v1.99.10-200-gd8f3120"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-200"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-da78428ac543d6d9ac2493dfa77364eb300a47cb #1 Mon Oct 23 15:28:01 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald 
[JSON_inventory_logging.txt](https://github.com/openbmc/openbmc/files/1415758/JSON_inventory_logging.txt)

gkeishin, 2017-10-25:
Seems to go away on the second reset though.

vishwabmc, 2017-10-26:
@gkeishin These are as a result of trying to de-serialize a non-versioned data from a versioned Cereal code and this is expected the `FIRST` time.  You must not see this in the below cases:

- 1/ After BMC reset
- 2/ Updating to a N+1 code level.

If you see the issue in those cases, then please re-open this.

closed 2017-10-26 by gkeishin