openbmc/openbmc #2518: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Failed with result 'core-dump

gkeishin, 2017-10-25:
`Journald snippet`
```
Oct 25 15:11:59 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=dumped, status=6/ABRT
Oct 25 15:11:59 witherspoon phosphor-fan-control[2610]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Oct 25 15:11:59 witherspoon phosphor-fan-control[2610]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Oct 25 15:11:59 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Unit entered failed state.
Oct 25 15:11:59 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Failed with result 'core-dump'
```

`Service status`
```
● phosphor-fan-control-init@witherspoon-fan-watchdog.service - Phosphor Fan Control Initialization
   Loaded: loaded (/lib/systemd/system/phosphor-fan-control-init@.service; disabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Wed 2017-10-25 15:07:30 UTC; 13min ago
  Process: 1894 ExecStart=/usr/bin/env phosphor-fan-control --init (code=dumped, signal=ABRT)
 Main PID: 1894 (code=dumped, signal=ABRT)

Oct 25 15:07:29 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Main process exited, code=dumped, status=6/ABRT
Oct 25 15:07:29 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Unit entered failed state.
Oct 25 15:07:29 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Failed with result 'core-dump'.
Oct 25 15:07:30 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Service hold-off time over, scheduling restart.
Oct 25 15:07:30 witherspoon systemd[1]: Stopped Phosphor Fan Control Initialization.
Oct 25 15:07:30 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Start request repeated too quickly.
Oct 25 15:07:30 witherspoon systemd[1]: Failed to start Phosphor Fan Control Initialization.
Oct 25 15:07:30 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Unit entered failed state.
Oct 25 15:07:30 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Failed with result 'core-dump'.
```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-113-g65edf7d-r5-5-g2521702"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-113-g65edf7d-r5"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-da78428ac543d6d9ac2493dfa77364eb300a47cb #1 Fri Oct 20 16:40:38 CDT 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-4f33e0f4/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.109
        op-build-v1.7-1507-g7d34499
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9-rc4
        hostboot-e8c85bb
        linux-4.13.7-openpower1-p76f6be3
        petitboot-v1.6.1-p7ea14c4
        machine-xml-eee619a
        occ-cff91e5
        hostboot-binaries-edd7653
        capp-ucode-p9-dd2-v2
        sbe-ec9a99e
root@witherspoon:~#
```

- [x] FFDC journald log 
[FAN_INIT_COREDUMP_IPL.txt](https://github.com/openbmc/openbmc/files/1415982/FAN_INIT_COREDUMP_IPL.txt)

spinler, 2017-10-25:
The issue was that the fan4_target file wasn't created in sysfs by the max31785 device driver, so it wasn't put on D-Bus, and then fan control would fail. 

There are changes coming (https://gerrit.openbmc-project.xyz/#/c/7516/ and https://gerrit.openbmc-project.xyz/#/c/7517/) that will move fan-control init to a 'wants' dependency instead of 'requires' so that the system will be able to boot in the future, and the fan watchdog will trip and just run all the fans at high speed.

spinler, 2017-10-25:
@amboar  It sounds like you may have a fix in the device driver.  Assigning to you.

msbarth, 2017-10-25:
The sysfs entries show that fan4_target is missing along with fan4's second rotor input(fan8_input).
```
root@witherspoon:/sys/class/hwmon/hwmon9# cat name
max31785a
root@witherspoon:/sys/class/hwmon/hwmon9# ls fan*
fan1_alarm   fan1_target  fan2_input   fan3_fault   fan4_input   fan7_input
fan1_fault   fan2_alarm   fan2_target  fan3_input   fan5_input
fan1_input   fan2_fault   fan3_alarm   fan3_target  fan6_input
root@witherspoon:/sys/class/hwmon/hwmon9# tail -v fan*_target
==> fan1_target <==
10500

==> fan2_target <==
10500

==> fan3_target <==
10500
root@witherspoon:/sys/class/hwmon/hwmon9# tail -v fan*_input
==> fan1_input <==
0

==> fan2_input <==
0

==> fan3_input <==
0

==> fan4_input <==
0

==> fan5_input <==
0

==> fan6_input <==
0

==> fan7_input <==
0
```

This would point to issue in the max31785 device driver presenting all of the sysfs entries to hwmon. Fan control init is configured for witherspoon to set fan1-4 target speeds to 10500rpm at poweron. When attempting to set the fan3_0 dbus target owned by hwmon9, an exception was thrown due to this object was never created causing fan control init to core. Fan control init is expected to core by design in this case currently. A change (https://gerrit.openbmc-project.xyz/#/c/7516/ & https://gerrit.openbmc-project.xyz/#/c/7517/) will alter fan control init's behavior to allow the system to still poweron, where in  this case, the watchdog should trigger the fans to full speed.

amboar, 2017-10-26:
I've [sent a series to the list](http://patchwork.ozlabs.org/project/openbmc/list/?series=10264&state=%2A&archive=both) and [posted a test commit to generate a build](https://gerrit.openbmc-project.xyz/#/c/7622/) that people can test out.

@eddiejames @spinler Can you please review the patches on the list? @gkeishin (and others) can you please test the CI build?

gkeishin, 2017-10-26:
```
------------
Driver Info
------------
Gerrith build : 
https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/6912/distro=ubuntu,target=witherspoon/artifact/deploy/images/witherspoon/obmc-phosphor-image-witherspoon-20171026063335.ubi.mtd.tar

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-175"
VERSION_ID="v1.99.10-175-g3c6e691"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-175"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
caLinux witherspoon 4.10.17-da78428ac543d6d9ac2493dfa77364eb300a47cb #1 Thu Oct 26 06:39:44 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/timestamp
20171026064102
root@witherspoon:~# cat /media/pnor-ro-4f33e0f4/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.109
        op-build-v1.7-1507-g7d34499
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9-rc4
        hostboot-e8c85bb
        linux-4.13.7-openpower1-p76f6be3
        petitboot-v1.6.1-p7ea14c4
        machine-xml-eee619a
        occ-cff91e5
        hostboot-binaries-edd7653
        capp-ucode-p9-dd2-v2
        sbe-ec9a99e
root@witherspoon:~#

-----------------
Test Ran 
-----------------

- [x] Code update - PASSED
- [x] Basic Power On - PASSED
- [x] HW CI test - PASSED
- [x] OS Reboot Test - PASSED
- [x] OS shutdown Test - PASSED 
- [x] Reset reload - Mixed result but doesnt crash anything as such, but Boot good. 

--------------------------------------------------------------
NOTE: This change doesnt have the MTU fix from Andrew Geissler
--------------------------------------------------------------

root@witherspoon:~# ifconfig eth0 | grep MTU
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
root@witherspoon:~#

--------------------------------------
Others notable warnings on journald: 
-------------------------------------

Seeing such logs on the journald, does this means something in this context of the fix.

( max31785 3-0052: Failed to set page 21, performed one-shot retry with success: 0)

root@witherspoon:~# dmesg | grep -e net -e max31785
[    1.810000] ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:23:9f from chip
[    1.810000] ftgmac100 1e660000.ethernet: Using NCSI interface
[    1.810000] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
[    2.160000] Driver for 1-wire Dallas network protocol.
[   14.840000] console [netcon0] enabled
[   14.840000] netconsole: network logging started
[   52.970000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 1
[   52.970000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 2
[   52.970000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[... 23 lines cut ...]
--------------------------------------------------
phosphor-discover-system-state@0.service: Failed 
-------------------------------------------------
Seen during HW CI power on 

root@witherspoon:~# systemctl status phosphor-discover-system-state@0.service | cat
● phosphor-discover-system-state@0.service - Reboot If Enabled
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.wants/../phosphor-discover-system-state@.service; disabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Sat 2017-10-21 03:26:37 UTC; 1h 1min ago
  Process: 1566 ExecStart=/usr/bin/env phosphor-discover-system-state --host %i (code=dumped, signal=ABRT)
 Main PID: 1566 (code=dumped, signal=ABRT)

Oct 21 03:26:09 witherspoon systemd[1]: Starting Reboot If Enabled...
Oct 21 03:26:31 witherspoon phosphor-discover-system-state[1566]: Host power is off, checking power policy
Oct 21 03:26:31 witherspoon phosphor-discover-system-state[1566]: power_policy=RESTORE, restoring last state
Oct 21 03:26:35 witherspoon phosphor-discover-system-state[1566]: Error in mapper call
Oct 21 03:26:35 witherspoon phosphor-discover-system-state[1566]: terminate called after throwing an instance of 'std::runtime_error'
Oct 21 03:26:35 witherspoon phosphor-discover-system-state[1566]:   what():  Error in mapper call
Oct 21 03:26:37 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Main process exited, code=dumped, status=6/ABRT
Oct 21 03:26:37 witherspoon systemd[1]: Failed to start Reboot If Enabled.
Oct 21 03:26:37 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Unit entered failed state.
Oct 21 03:26:37 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Failed with result 'core-dump'.
root@witherspoon:~#

```

amboar, 2018-04-16:
We have work-arounds in place now

closed 2018-04-16 by amboar