openbmc/openbmc #2519: OpenBMC slow to respond to IPMI requests.

sammj, 2017-10-26:
On recent builds we've noticed that OpenBMC can take >= 5 seconds to respond to IPMI requests such as "get boot options" and "get device ID". From the OPAL log:

```
[   25.128211367,7] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[   25.128217620,7] BT: seq 0x0f netfn 0x00 cmd 0x09: Message sent to host
[   25.272654465,7] BT: seq 0x0f netfn 0x00 cmd 0x09: IPMI MSG done
[   25.272664328,7] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x07)
   (completed in 0.15 seconds)
[   25.272697415,7] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[   25.272704531,7] BT: seq 0x10 netfn 0x00 cmd 0x09: Message sent to host
[   30.296246358,7] BT: seq 0x10 netfn 0x00 cmd 0x09: IPMI MSG done
[   30.296249357,7] IPMI: Got error response 0xce
[   30.296253888,7] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x00)
   (completed in 5.02 seconds)
[   30.296262753,7] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[   30.296269226,7] BT: seq 0x11 netfn 0x00 cmd 0x09: Message sent to host
[   35.379288918,7] BT: seq 0x11 netfn 0x00 cmd 0x09: IPMI MSG done
[   35.379292133,7] IPMI: Got error response 0xce
[   35.379296641,7] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x00)
   (completed in 5.08 seconds)
[   35.379305813,7] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[   35.379311946,7] BT: seq 0x12 netfn 0x00 cmd 0x09: Message sent to host
[   40.433386533,7] BT: seq 0x12 netfn 0x00 cmd 0x09: IPMI MSG done
[   40.433389101,7] IPMI: Got error response 0xce
[   40.433392933,7] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x00)
   (completed in 5.06 seconds)
````

Which we worked around in Petitboot by reflecting the 10 second timeout OpenBMC advertises: https://github.com/open-power/petitboot/commit/093178c4a93ab6f25aa40101de1910368260228f

However having to wait for OpenBMC to do whatever it is doing can delay boot by 20-40 seconds since Petitboot makes several IPMI requests on init.
OpenBMC certainly seems capable of responding faster (see the first request/response above), what is happening in later requests to cause this large delay?

dkodihal, 2017-10-27:
Is this being observed amidst a boot, or at any point of time?

tomjoseph83, 2017-10-27:
@sammj In a good case, i have observed that Petitboot request for the System Boot Options 4 times both for parameter 0x05 and  0x61 What is the rationale behind that, is there an inefficiency?

```
IPMI GET_SYS_BOOT_OPTIONS
IPMI Response:
0x000000: 00 01 05 80 0c 00 00 00                         ........        
IPMI Incoming: Seq 0x10, NetFn 0x00, CMD: 0x09 
0x000000: 61 00 00                                        a..             
IPMI GET_SYS_BOOT_OPTIONS
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
IPMI Response:
0x000000: ff 01 61 00 00 00 00 00 00 00 00 00 00 00 00 00 ..a.............
0x000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000030: 00 00 00                                        ...             
IPMI Incoming: Seq 0x11, NetFn 0x00, CMD: 0x09 
0x000000: 05 00 00                                        ...             
IPMI GET_SYS_BOOT_OPTIONS
IPMI Response:
0x000000: 00 01 05 80 0c 00 00 00                         ........        
IPMI Incoming: Seq 0x12, NetFn 0x00, CMD: 0x09 
0x000000: 61 00 00                                        a..             
IPMI GET_SYS_BOOT_OPTIONS
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
IPMI Response:
0x000000: ff 01 61 00 00 00 00 00 00 00 00 00 00 00 00 00 ..a.............
0x000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000030: 00 00 00                                        ...             
IPMI Incoming: Seq 0x13, NetFn 0x00, CMD: 0x09 
0x000000: 05 00 00                                        ...             
IPMI GET_SYS_BOOT_OPTIONS
IPMI Response:
0x000000: 00 01 05 80 0c 00 00 00                         ........        
IPMI Incoming: Seq 0x14, NetFn 0x00, CMD: 0x09 
0x000000: 61 00 00                                        a..             
IPMI GET_SYS_BOOT_OPTIONS
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
IPMI Response:
0x000000: ff 01 61 00 00 00 00 00 00 00 00 00 00 00 00 00 ..a.............
0x000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000030: 00 00 00                                        ...             
IPMI Incoming: Seq 0x15, NetFn 0x00, CMD: 0x09 
0x000000: 05 00 00                                        ...             
IPMI GET_SYS_BOOT_OPTIONS
IPMI Response:
0x000000: 00 01 05 80 0c 00 00 00                         ........        
IPMI Incoming: Seq 0x16, NetFn 0x00, CMD: 0x09 
0x000000: 61 00 00                                        a..             
IPMI GET_SYS_BOOT_OPTIONS
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
IPMI Response:
0x000000: ff 01 61 00 00 00 00 00 00 00 00 00 00 00 00 00 ..a.............
0x000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x000030: 00 00 00                                  
```

amboar, 2017-10-29:
@tomjoseph83 please create a separate issue against petitboot for your observed inefficiency. This issue is to investigate why OpenBMC can take up to 5 seconds to respond to IPMI queries.

mine260309, 2017-10-30:
Please check `iBTCR0` status from BMC:
```
devmem 0x1e789140
```
The expected value should be `0x00E4A00D`, if it becomes `0x00E4000D` (e.g. after ipmi mc reset), it means the register is reset and it's the root cause.

If that is the case, it's known issue and we had discussion with @jk-ozlabs and it's more likely to be fixed in host side.

sammj, 2017-11-15:
I ran this test on wsbmc008 today,

**Before**
```
root@witherspoon:~# devmem 0x1e789140
0x00E4A00D
```
**After**
```
root@witherspoon:~# devmem 0x1e789140
0x00E4A00D
```

And we can see from the Skiboot log that requests are taking multiple seconds:
```
[  277.350097217,8] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[  277.350106995,6] BT: seq 0x1a netfn 0x00 cmd 0x09: Message sent to host
[  278.255660550,6] BT: seq 0x1a netfn 0x00 cmd 0x09: IPMI MSG done
[  278.255667563,8] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x07)
[  278.255694863,8] opal_ipmi_send(cmd: 0x09 netfn: 0x00 len: 0x03)
[  278.255701020,6] BT: seq 0x1b netfn 0x00 cmd 0x09: Message sent to host
[  282.475547782,6] BT: seq 0x1b netfn 0x00 cmd 0x09: IPMI MSG done
[  282.475550660,7] IPMI: Got error response 0xff
[  282.475554866,8] opal_ipmi_recv(cmd: 0x09 netfn: 0x01 resp_size: 0x32)
[  282.475641379,8] opal_ipmi_send(cmd: 0x02 netfn: 0x0c len: 0x04)
[  282.475647357,6] BT: seq 0x1c netfn 0x0c cmd 0x02: Message sent to host
[  284.057356850,6] BT: seq 0x1c netfn 0x0c cmd 0x02: IPMI MSG done
[  284.057361480,8] opal_ipmi_recv(cmd: 0x02 netfn: 0x0d resp_size: 0x07)
[  284.057382643,8] opal_ipmi_send(cmd: 0x01 netfn: 0x06 len: 0x00)
[  284.057387351,6] BT: seq 0x1d netfn 0x06 cmd 0x01: Message sent to host
[  284.174332574,6] BT: seq 0x1d netfn 0x06 cmd 0x01: IPMI MSG done
[  284.174336960,8] opal_ipmi_recv(cmd: 0x01 netfn: 0x07 resp_size: 0x0f)
[  284.174364653,8] opal_ipmi_send(cmd: 0x1a netfn: 0x3a len: 0x00)
[  284.174369352,6] BT: seq 0x1e netfn 0x3a cmd 0x1a: Message sent to host
[  284.182193991,6] BT: seq 0x1e netfn 0x3a cmd 0x1a: IPMI MSG done
[  284.182196476,7] IPMI: Got error response 0xc1
```

So at first impression this doesn't appear to be due to a register reset.

closed 2019-10-27 by stale[bot]