openbmc/openbmc #2521: Palmetto: NCSI network broken on host power on/off

mine260309, 2017-10-26:
Palmetto BMC's network breaks when host is powered on/off if both network ports are connected.
This issue exists for a long time.

### Background
* Palmetto BMC uses NCSI connected to BCM5718 chip, with two ports, say port0, port1
* Prepare two LANs, say LAN1 (e.g. 192.168.xxx), LAN2 (9.123.xxx);
* Assign static LAN1 IP to BMC (e.g. 192.168.0.60), and expect Host to get LAN2's IP.
### Steps
1. Connect port0 to LAN1;
2. AC on, BMC network is OK in LAN1;
3. Before powering on host, connect port1 to LAN2, it does not break anything;
4. Power on the host, after host (e.g. Petiboot or OS) is running, it will reset the PCIE devices;
Now BMC network is down. while it is expected to work.
5. Disconnect port1 (LAN2), BMC network is working again.

### Additional info (switch the ports)
6. Disconnect port0 (LAN1), BMC network is down, which is expected;
7. Connect port1 to LAN1, BMC network is still down, while it is expected to work;
8. Connect port0 to LAN2, BMC network is still down;
9. Disconnect port0 (LAN2), BMC network is up and working fine.
10. Connect port0 to LAN2 again, BMC network is still working.

The issue is related to step 4, 7 and 8, where the BMC network is expected to be working.

mine260309, 2017-10-26:
@sammj Please have a look. It should be related to issues about NCSI selecting channels.

mine260309, 2017-10-27:
Some information:
A vendor meet the same issue, and they have a workaround in ncsi driver to always use channel 0 for BMC. This resolves the issue but the side effect is that only port0 can be used for BMC.

sammj, 2017-10-27:
Yeah I wondered if some people were just hacking it to channel 0. As the NCSI driver stands today what's happening in your steps is:
- NCSI selects first channel with link - channel 0
- Host power on triggers an AEN event, channel 0 is briefly down and NCSI switches to channel 1
- Disconnecting port 1 triggers NCSI to switch back to channel 0 which has link again.

NCSI doesn't know anything about how the network is configured, only that a channel has link, which is why you see it stay on channel 1 even though you obviously can't route 192.168.xxx.xxx on the 9.123.xxx.xxx network.
The non-hacky way of working around this to provide a way for userspace to a) query what channels are available and b) tell the NCSI driver to preference a certain channel if desired. I'm looking in to adding this functionality.

mine260309, 2017-10-27:
@sammj Thanks for the explanation.

What about the other case:
1. Connect port0 to LAN1;
2. AC on, BMC network is OK in LAN1;
3. Disconnect port0 (LAN1), BMC network is down, expected;
4. Connect port1 to LAN1, BMC network is still down.

In step 4, I expect NCSI driver to select channel1 because channel0 is down and channel1 is up, but it's not working.

sammj, 2017-10-27:
Yes that's odd. Does this occur with the latest NCSI patches?
I'll try that out on a local machine and see if it recreates.

gtmills, 2018-08-20:
@mine260309 @sammj Still an issue? Or can this be closed?

sammj, 2018-08-23:
A number of changes have gone into the NCSI driver that would affect this scenario - worth retesting if someone finds the time on a Palmetto.

closed 2019-03-28 by mine260309