openbmc/openbmc #2524: phosphor-mboxd: Flash read where there is no partition error handling support.

ojayanth, 2017-10-27:
The requirements are two fold:

Refer #2365 for background on this issue.
Comments from @amboar  
   " The behaviour between virtualised and non-virtualised PNORs from the perspective of the host should be as similar as possible. Currently this is violated on a number of fronts, including this one. Anything otherwise needs to be documented and discussed with the hostboot and OPAL communities before being implemented and rolled out
    The flash should not become inaccessible due to plausible access from the host. This is a denial of service vector. The host must not be able to crash processes on the BMC"

sjitindarsingh, 2017-10-27:
I think we need to consider the desired behaviour when the host tries to read outside the partition limits. I think the expected result from the host side is that it should work (shouldn't return an error). Since mbox is designed to be a flash abstraction, not a pnor abstraction, there is no reason to fail a read outside a partition if it is still within the limits of flash. If you are going to return an error then you need to modify the side of flash which is being reported to the host so that it matches the pnor size.

amboar, 2017-11-17:
This should be fixed sooner rather than later as it affects Ubuntu's behaviour on the host, which scans mtd_block devices for filesystem detection purposes.

amboar, 2018-03-28:
The problem has been resolved for the read case by https://github.com/openbmc/phosphor-mboxd/commit/ae1edb940c65e13e610108203e6a1cba0ab424c7. Writes are a whole other matter.

closed 2018-03-28 by amboar