openbmc/openbmc #2525: Sprint 17.11.01 - Committed

rfrandse, 2017-10-27:


dkodihal, 2017-10-30:
**Squad 1 sprint plan (WIP)**

**Ratan**
<Bug fix>
Wrap up event-logging - #2254 and #2399
Enhance networking UT, especially VLAN (work with Nagaraju if need be)
Create event for n/w changes - need new enhancement issue.
Add n/w specific plug-in to user initiated dump - need new enhancement issue.
Review networking tests with FVT.
Networking doc updates.

**Jayanth**
<bug fix>
dreport - use getManagedObjects for better performance - need new issue.
Dump - collect inventory - need new issue.
ffdc script - call dreport - need new issue.
dump performance issue - defect open.
Cereal 0-length file issue - long term fix (Dhruv could help) - issue already exists.

**Dhruv & Tom**
<bug fix>
ipmi log improvement - need new issue
sensors UT/test plan with FVT
ipmid performance during IPL - issues exist

**Devender**
<bug fix>
Error log UT and doc

**Nagaraju**
<bug fix>
op-test-framework : switch to xyz interfaces
Help Ratan with n/w and Devender with logging

**Deepak**
Fix-up InternalErrors to ensure metadata exists - need issue
Event notification client examples and test plan
UT and doc

**Naveen and Aditya**
<bug fixes>

spinler, 2017-10-31:
***Squad 2 Sprint Plan***

**Matt S**

**Matt B**

**Gunnar**

**Brandon**

**Eddie**

closed 2017-11-30 by rfrandse