openbmc/openbmc #2529: OpenBMC : Witherspoon : BMC is stuck at 'lm75 9-004a: hwmon0: sensor 'tmp275'' step after BMC reboot

cdeadmin, 2017-10-27:


eddiejames, 2017-10-30:
BMC will hang during boot if chassis power is on, but host is not started.

This is because the SBE is not running. The occ-hwmon driver loads during BMC boot and attempts to probe it's platform devices (they are in the device tree). occ-hwmon probe accesses the SBEFIFO driver, which starts a transfer and waits for a response. It can never receive one, since the SBE isn't running, and there is no timeout.

I have a fix coded up which will timeout the SBEFIFO driver waiting for a response. The BMC seems to continue booting after 870 seconds, due to an FSI operation failure, meaning that the SBEFIFO driver will fail out instead of continuing to hang.

rfrandse, 2017-11-02:
Fixed in -r9

closed 2017-11-02 by rfrandse

amboar, 2017-11-13:
@rfrandse `-r9` refers to an IBM internal tag. Please reference publicly accessible commits and tags on the public issue tracker.