openbmc/openbmc #2530: pypi.python.org now requires https for downloads

geissonator, 2017-10-27:
Build team hit this one today, hit's 2 of our packages:
```
--2017-10-27 13:45:48--  http://pypi.python.org/packages/source/g/gevent/gevent-1.0.1.tar.gz
Resolving pypi.python.org... 151.101.20.223, 2a04:4e42:5::223
Connecting to pypi.python.org|151.101.20.223|:80... connected.
HTTP request sent, awaiting response... 403 SSL is required
2017-10-27 13:45:48 ERROR 403: SSL is required.

--2017-10-27 13:45:48--  http://pypi.python.org/packages/source/I/IPy/IPy-0.83.tar.gz
Resolving pypi.python.org... 151.101.20.223, 2a04:4e42:5::223
Connecting to pypi.python.org|151.101.20.223|:80... connected.
HTTP request sent, awaiting response... 403 SSL is required
2017-10-27 13:45:48 ERROR 403: SSL is required.
```

A lot of us don't hit it because we have the package cached.  https://github.com/pypa/pip/issues/4817#issuecomment-339906906 seems to discuss the issue and the reasoning.  I'll be applying a couple fixes to get this going again.

rfrandse, 2017-10-27:
<https://gerrit.openbmc-project.xyz/7650> Update pypi.python.org packages to use https
Resolves  #2530 pypi.python.org now requires https for downloads

closed 2017-10-30 by geissonator

geissonator, 2017-10-31:
Official upstream fix coming in via https://lists.yoctoproject.org/pipermail/meta-virtualization/2017-October/002932.html

mine260309, 2018-03-22:
The fix is reverted at 36af49f, but upstream fix has not been merged (as of 8280d07)
So the issue shall be re-opened and I do get feedback from system vendors that they get build failure on this.
@geissonator @bradbishop

geissonator, 2018-03-22:
Brad merged https://gerrit.openbmc-project.xyz/#/c/9586/ which should fix this issue (again).