openbmc/openbmc #2531: phosphor-hwmon: update threshold checking

mine260309, 2017-10-30:
This task if from https://github.com/openbmc/openbmc/issues/2387#issuecomment-338539642
That the threshold value checking shall be changed to >= or <=, intead of > or <.

Reasons:
1. According to IPMI Spec "35.13.3 High-going versus Low-going Threshold Events":
> A high-going threshold has its assertion events become set when the reading is >= the threshold, while for a low-going event the assertion event becomes set when the reading is <= the threshold.
2. For example for the fan speeds, the lowest value is 0, and if user wants a warning on this case, it has to be <=, the current code "<" will cause the warning never fired.

mine260309, 2017-10-30:
@chiabinglee will help to fix this.

rfrandse, 2017-10-30:
<https://gerrit.openbmc-project.xyz/7663> Update threshold checking comparison operators
Resolves  #2531 phosphor-hwmon: update threshold checking

closed 2017-11-03 by geissonator