openbmc/openbmc #2532: log entries are not getting restored after code update.

ngorugan, 2017-10-30:


amboar, 2017-10-30:
@ngorugan Please provide a description for this issue detailing the versions affected and any evidence to support the subject. Otherwise we have no idea whether this is relevant in the future.

ngorugan, 2017-11-03:
```
Driver info
 ```
```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-64929c7fd2d88ef6a9e18d6a484ddc3957a6ecab #1 Thu Nov 2 12:34:59 CDT 2017 armv6l GNU/Linux
root@witherspoon:~# cat  /proc/version
Linux version 4.10.17-64929c7fd2d88ef6a9e18d6a484ddc3957a6ecab (bmcbuild@hmcci187.aus.stglabs.ibm.com) (gcc version 6.2.0 (GCC) ) #1 Thu Nov 2 12:34:59 CDT 2017
root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-113-g65edf7d-r10-0-gcdf7635"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-113-g65edf7d-r10"
```

ngorugan, 2017-11-03:
I have done code update using below given command
```
export BMC_IP=9.3.185.33

curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://$BMC_IP/login

curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" -X POST -T /gsa/ausgsa/projects/o/openbmc/obmc910/Builds/ob910.1744.20171102n/output/build/tmp/deploy/images/witherspoon/obmc-phosphor-image-witherspoon-20171102171257.ubi.mtd.tar  https://$BMC_IP/upload/image

curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP/xyz/openbmc_project/software/enumerate

curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":"xyz.openbmc_project.Software.Activation.RequestedActivations.Active"}' https://$BMC_IP/xyz/openbmc_project/software/efc8a851/attr/RequestedActivation

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST  -d '{"data": []}' https://$BMC_IP/xyz/openbmc_project/software/efc8a851/action/delete
```

```
Console o/p
```
```
root@witherspoon:..# cat  /proc/version
Linux version 4.10.17-67fda6b3a8b85e3063fc46fb3262ffb130ba396d (ngorugan@gfw161.aus.stglabs.ibm.com) (gcc version 6.2.0 (GCC) ) #1 Thu Nov 2 00:24:02 CDT 2017
root@witherspoon:..# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-1"
VERSION_ID="v2.0-1-g39803a5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-1"
BUILD_ID="v2.0"
root@witherspoon:..# busctl tree xyz.openbmc_project.Logging  | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/logging
      ├─/xyz/openbmc_project/logging/entry
      │ ├─/xyz/openbmc_project/logging/entry/12
      │ └─/xyz/openbmc_project/logging/entry/13
      └─/xyz/openbmc_project/logging/internal
        └─/xyz/openbmc_project/logging/internal/manager
root@witherspoon:..#
root@witherspoon:..#
root@witherspoon:..#
root@witherspoon:..# shutdown -r now
[  OK  ] Stopped Assert bmc_booted LED.
[  OK  ] Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
[  OK  ] Stopped target Poweri Off.
[  OK  ] Removed slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped Wait for /xyz/openbmc_project/control/host0.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2drunning.slice.
[  OK  ] Stopped target Host0 running after reset.
[  OK  ] Removed slice system-phosphor\x2dcooling\x2dtype.slice.
[  OK  ] Removed slice system-obmc\x2dled\x2dgroup\x2dstart.slice.
[  OK  ] Stopped target Stop Host0.
[  OK  ] Removed slice system-phosphor\x2dmax31785\x2dmsl.slice.
         Stopping Phosphor sysfs LED controller...
Nov 03 17:55:06 witherspoon systemd[1]: Stopped Assert bmc_booted LED.
Nov 03 17:55:06 witherspoon systemd[1]: Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
Nov 03 17:55:06 witherspoon systemd[1]: Stopped target Poweri Off.
Nov 03 17:55:06 witherspoon systemd[1]: Removed slice system-obmc\x2dflash\x2dbmc\x2dupdateubootvars.slice.
Nov 03 17:55:06 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Nov 03 17:55:06 witherspoon systemd[1]: Stopped Wait for /xyz/openbmc_project/control/host0.
Nov 03 17:55:06 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Nov 03 17:55:06 witherspoon systemd[1]: Removed slice system-op\x2dreset\x2dchassis\x2drunning.slice.
Nov 03 17:55:06 witherspoon systemd[1]: Stopped target Host0 running after reset.
Nov 03 17:55:06 witherspoon systemd[1]: Removed slice system-phosphor\x2dcooling\x2dtype.slice.
Nov 03 17:55:06 witherspoon systemd[1]: Removed slice system-obmc\x2dled\x2dgroup\x2dstart.slice.
Nov 03 17:55:06 witherspoon systemd[1]: Stopped target Stop Host0.
Nov 03 17:55:06 witherspoon systemd[1]: Removed slice system-phosphor\x2dmax31785\x2dmsl.slice.
Nov 03 17:55:06 witherspoon systemd[1]: Stopping Phosphor sysfs LED controller...
Nov 03 17:55:06 witherspoon systemd[1]: Stopped Assert bmc_booted LED.
[... 958 lines cut ...]
[   94.170000] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" started, PID 1184

root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~# busctl tree xyz.openbmc_project.Logging  | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/logging
      └─/xyz/openbmc_project/logging/internal
        └─/xyz/openbmc_project/logging/internal/manager
root@witherspoon:~# [  100.550000] UBIFS (ubi8:3): un-mount UBI device 8
[  100.550000] UBIFS (ubi8:3): background thread "ubifs_bgt8_3" stops

root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~#
root@witherspoon:~# [  134.820000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[  135.040000] 8021q: adding VLAN 0 to HW filter on device eth0
[  136.200000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[  204.270000] max31785 3-0052: Failed to set page 5, performed one-shot retry with success: 0

```

dkodihal, 2017-11-06:
@ngorugan can you also check if before the reboot error logs were persisted under /var/lib/phosphor-logging, and also did they exist after the reboot? @anoo1 it seems like the error logs were lost after the reboot post code update.

ngorugan, 2017-11-09:
Today I tried to recreate the issue with below given drivers, but not able to reproduce.
```ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r0-0-g9f22e7c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r0"

root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r2-0-g4b46995"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r2"

root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-1"
VERSION_ID="v2.0-1-g39803a5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-1"
BUILD_ID="v2.0"

root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-1"
VERSION_ID="v2.0-1-g39803a5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-1"
BUILD_ID="v2.0"
````

ngorugan, 2017-11-09:
Before I reject this issue want to test one more time.

ngorugan, 2017-11-20:
Today have tested with below given drivers, but not able to reproduce.

```root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r7-0-ge521a31"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r7"
root@witherspoon:~#```

```root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r9-0-gecda565"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r9"
root@witherspoon:~#```

```root@witherspoon:~# cat  /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-52"
VERSION_ID="v2.0-52-g158f8de"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-52"
BUILD_ID="v2.0" ```

closed 2017-11-20 by ngorugan