openbmc/openbmc #2536: phosphor-hwmon-read: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error

gkeishin, 2017-10-31:
```
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2519]: Usage: phosphor-hwmon-readd [options]
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2519]: Options:
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2519]:     --help               print this menu
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2519]:     --path=<path>        sysfs location to monitor
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2519]:     --of-name=<path>     open firmware device name to monitor
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2522]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Oct 25 22:56:26 witherspoon phosphor-hwmon-readd[2522]:   what():  filesystem error: cannot canonicalize: No such file or directory [/sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a180.i2c-bus/i2c-5/5-0070/driver] [/]
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]: Usage: phosphor-hwmon-readd [options]
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]: Options:
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]:     --help               print this menu
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]:     --path=<path>        sysfs location to monitor
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]:     --of-name=<path>     open firmware device name to monitor
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2516]: ERROR: Path not specified or invalid.
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2519]: ERROR: Path not specified or invalid.
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]: Usage: phosphor-hwmon-readd [options]
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]: Options:
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]:     --help               print this menu
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]:     --path=<path>        sysfs location to monitor
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]:     --of-name=<path>     open firmware device name to monitor
Oct 25 22:56:27 witherspoon phosphor-hwmon-readd[2525]: ERROR: Path not specified or invalid.
Oct 25 22:56:27 witherspoon kernel: max31785 3-0052: Failed to set page 10, performed one-shot retry with success: 0
Oct 25 22:56:27 witherspoon kernel[984]: [  478.050000] max31785 3-0052: Failed to set page 10, performed one-shot retry with success: 0
Oct 25 22:56:27 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Oct 25 22:56:27 witherspoon avahi-daemon[985]: Service "sftp #13" (/services/sftp.service) successfully established.
Oct 25 22:56:27 witherspoon kernel[984]: [  478.520000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Oct 25 22:56:28 witherspoon mboxctl[2537]: Reset: Success
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Main process exited, code=exited, status=255/n/a
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit entered failed state.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Failed with result 'exit-code'.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Main process exited, code=exited, status=255/n/a
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit entered failed state.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Failed with result 'exit-code'.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Main process exited, code=dumped, status=6/ABRT
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit entered failed state.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Failed with result 'core-dump'.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Main process exited, code=exited, status=255/n/a
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit entered failed state.
Oct 25 22:56:28 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Failed with result 'exit-code'.
```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-233"
VERSION_ID="v1.99.10-233-g205fc37"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-233"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-64929c7fd2d88ef6a9e18d6a484ddc3957a6ecab #1 Mon Oct 30 14:22:46 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-7617ca43/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.113
        op-build-v1.7-1522-g977d38f
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9-rc5
        hostboot-510df2f
        linux-4.13.8-openpower2-pc35af39
        petitboot-v1.6.1-p1014504
        machine-xml-eee619a
        occ-7449acd
        hostboot-binaries-56bf0b7
        capp-ucode-p9-dd2-v2
        sbe-ec9a99e
root@witherspoon:~#
```

- [x] FFDC journald 
[HWMON_CORE_DUMP.txt](https://github.com/openbmc/openbmc/files/1429664/HWMON_CORE_DUMP.txt)

amboar, 2017-10-31:
Some of this might be attributed to a race between the application and the kernel as the host power-on process unbinds and binds the ir35221. That's a guess though, and someone with more knowledge in this area should take a look.

geissonator, 2017-10-31:
Yeah, pretty sure I see these occasionally with the race conditions on bind/unbind in the power on/off flow.  Anything we can do here @spinler or just ignore it?

spinler, 2017-10-31:
Yea, I think it is a race.  While we can ignore, it would be good to fix.  I will take a look.

spinler, 2017-11-08:
We are finally removing support for the 'P9 DD1 VCS workaround' on Witherspoon, which is the reason we are unbinding these drivers on power off.  So this problem will go away.

gkeishin, 2018-04-24:
Do we still need this or I can close this one up @spinler

gkeishin, 2018-04-24:
Closing after discussion

closed 2018-04-24 by gkeishin