openbmc/openbmc #2542: Journald log showing last reset data

gkeishin, 2017-10-31:
Code update the new Master driver https://openpower.xyz/job/openbmc-build/1084/

post update `reboot` BMC and we see stale old reset data in the journald no pager

```
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /media/rofs-d50a6a75.
Oct 30 22:57:37 witherspoon systemd[1]: etc-machine\x2did.mount: Mount process exited, code=exited status=1
Oct 30 22:57:37 witherspoon systemd[1]: Failed unmounting /etc/machine-id.
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /var/volatile.
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /media/pnor-rw-7617ca43.
Oct 30 22:57:37 witherspoon kernel: UBIFS (ubi8:3): un-mount UBI device 8
Oct 30 22:57:37 witherspoon kernel: UBIFS (ubi8:3): background thread "ubifs_bgt8_3" stops
Oct 30 22:57:37 witherspoon systemd[1]: Unmounting /etc...
Oct 30 22:57:37 witherspoon systemd[1]: Unmounting /var...
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /etc.
Oct 30 22:57:37 witherspoon systemd[1]: var.mount: Mount process exited, code=exited status=1
Oct 30 22:57:37 witherspoon systemd[1]: Failed unmounting /var.
Oct 30 22:57:37 witherspoon systemd[1]: media-rofs\x2d1db3fb1b.mount: Mount process exited, code=exited status=1
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /media/rofs-1db3fb1b.
Oct 30 22:57:37 witherspoon systemd[1]: media-rofs\x2d1db3fb1b.mount: Unit entered failed state.
Oct 30 22:57:37 witherspoon systemd[1]: usr-local-share-pnor.mount: Mount process exited, code=exited status=1
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /usr/local/share/pnor.
Oct 30 22:57:37 witherspoon systemd[1]: usr-local-share-pnor.mount: Unit entered failed state.
Oct 30 22:57:37 witherspoon systemd[1]: Unmounting /usr/local...
Oct 30 22:57:37 witherspoon systemd[1]: Unmounted /usr/local.
Oct 30 22:57:38 witherspoon kernel: UBIFS (ubi8:2): un-mount UBI device 8
Oct 30 22:57:38 witherspoon kernel: UBIFS (ubi8:2): background thread "ubifs_bgt8_2" stops
Oct 30 22:57:38 witherspoon systemd[1]: Unmounted /media/pnor-prsv.
Oct 30 22:57:38 witherspoon systemd[1]: Unmounted /media/pnor-ro-7617ca43.
Oct 30 22:57:38 witherspoon systemd[1]: Unmounted Temporary Directory.
Oct 30 22:57:38 witherspoon systemd[1]: Stopped target Swap.
Oct 30 22:57:38 witherspoon systemd[1]: Unmounting /media...
Oct 30 22:57:38 witherspoon systemd[1]: Unmounted /media.
Oct 30 22:57:38 witherspoon systemd[1]: Reached target Unmount All Filesystems.
Oct 30 22:57:38 witherspoon systemd[1]: Stopped target Local File Systems (Pre).
Oct 30 22:57:38 witherspoon systemd[1]: Stopped Remount Root and Kernel File Systems.
Oct 30 22:57:38 witherspoon systemd[1]: Stopped Create Static Device Nodes in /dev.
Oct 30 22:57:38 witherspoon systemd[1]: Reached target Shutdown.
Oct 30 22:57:38 witherspoon systemd[1]: Reached target Final Step.
Oct 30 22:57:38 witherspoon systemd[1]: Starting Reboot...
Oct 30 22:57:38 witherspoon systemd[1]: Shutting down.
Oct 30 22:57:38 witherspoon systemd[1]: Hardware watchdog 'aspeed_wdt', version 0
Oct 30 22:57:38 witherspoon systemd[1]: Set hardware watchdog to 10min.
Oct 30 22:57:38 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Oct 30 22:57:38 witherspoon kernel: systemd-shutdow: 66 output lines suppressed due to ratelimiting
Oct 30 22:57:39 witherspoon systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Oct 30 22:57:39 witherspoon systemd-journald[955]: Journal stopped
```
- [x] journald 
[SERIAL_CONSOLE_HANG.txt](https://github.com/openbmc/openbmc/files/1431971/SERIAL_CONSOLE_HANG.txt)

gkeishin, 2017-11-01:
This is probably due to time ownership setting, set wrongly as @geissonator  pointed out earlier in a discussion.  I ll check and close this one out

gkeishin, 2017-11-01:
```
URI:https://xx.xx.xx.xx/xyz/openbmc_project/time/owner/attr/TimeOwner, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.Time.Owner.Owners.BMC",
  "message": "200 OK",
  "status": "ok"
}
URI:https://xx.xx.xx.xx/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.Time.Synchronization.Method.NTP",
  "message": "200 OK",
  "status": "ok"
}
```

gkeishin, 2017-11-01:
```
root@witherspoon:~# hwclock --utc --systohc
root@witherspoon:~# hwclock --show
Mon Oct 30 22:40:56 2017  0.000000 seconds
root@witherspoon:~# date
Mon Oct 30 22:40:59 UTC 2017
root@witherspoon:~# uptime
 22:41:07 up 7 min,  load average: 0.37, 1.02, 0.75
root@witherspoon:~#
```
Not sure if RTC is broken here.. still showing older date

gkeishin, 2017-11-01:
above step seems to kind of solve the problem, but not sure how.

mdmillerii, 2017-11-01:
Not sure exactly what you are observing, but I see the "current" date `Mon Oct 30 22:40:59 UTC 2017` was before the log date `Oct 30 22:57:38`.  I don't expect `systemd` to support dates going backwards and I can see that extending to determining which log is the most recent.  May need to take this into account in how the time test cases interact with general testing.

geissonator, 2017-11-01:
Yeah, specifically on George's system yesterday I saw this:
```
Oct 31 18:11:32 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Oct 31 18:11:32 witherspoon kernel: systemd-shutdow: 69 output lines suppressed due to ratelimiting
Oct 31 18:11:34 witherspoon systemd-journald[954]: Journal stopped
[witherspoon]# date
Tue Oct 31 17:57:52 UTC 2017
```

The BMC had gone back in time after the reboot.  journalctl is def not going to be able to handle that :)  Persisting the journal on witherspoon has brought this one out as an issue.

@mine260309 did our RTC work if you have any questions in that area @gkeishin

gkeishin, 2017-11-02:
I see this today on another system w34
```
Nov 01 20:53:56 witherspoon systemd[1]: var.mount: Mount process exited, code=exited status=1
Nov 01 20:53:56 witherspoon systemd[1]: Failed unmounting /var.
Nov 01 20:53:57 witherspoon systemd[1]: Reached target Unmount All Filesystems.
Nov 01 20:53:57 witherspoon systemd[1]: Stopped target Local File Systems (Pre).
Nov 01 20:53:57 witherspoon systemd[1]: Stopped Create Static Device Nodes in /dev.
Nov 01 20:53:57 witherspoon systemd[1]: Stopped Remount Root and Kernel File Systems.
Nov 01 20:53:57 witherspoon systemd[1]: Reached target Shutdown.
Nov 01 20:53:57 witherspoon systemd[1]: Reached target Final Step.
Nov 01 20:53:57 witherspoon systemd[1]: Starting Reboot...
Nov 01 20:53:57 witherspoon systemd[1]: Shutting down.
Nov 01 20:53:58 witherspoon systemd[1]: Hardware watchdog 'aspeed_wdt', version 0
Nov 01 20:53:58 witherspoon systemd[1]: Set hardware watchdog to 10min.
Nov 01 20:53:58 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Nov 01 20:53:58 witherspoon kernel: systemd-shutdow: 69 output lines suppressed due to ratelimiting
Nov 01 20:53:58 witherspoon systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Nov 01 20:53:58 witherspoon systemd-journald[953]: Journal stopped
root@witherspoon:~# date
Wed Nov  1 20:48:57 UTC 2017
root@witherspoon:~#
```
Looks like it doesn shows up..

gkeishin, 2017-11-02:
Time for someone to debug here.. I got bunch of system hitting this one.

gkeishin, 2017-11-02:
@mine260309 Can u let us know what data u ll need if it needs to debug from HW clock point of view.. Thanks

geissonator, 2017-11-02:
Retitled issue to reflect the problem.  Basically, after a BMC reboot, the "date" on the BMC can go backwards in time.  Some systems work fine though, so the theory is maybe bad RTC hardware or battery?  If you run 
```
hwclock --utc --systohc
hwclock --show
```

Then I believe things are ok, even after a BMC reboot, until you PDU the system.  @gkeishin can provide more details.

mdmillerii, 2017-11-02:
systemd has a "feature" to forward a clock to a date stamp on bootup.  If the timestamp is ahead of real time then resetting to ntp will repeatedly fail.

mine260309, 2017-11-03:
I do not recreate this issue on w34 with PDU cycle, the hwclock is fine on it.
But I do hit this issue on `WS-Seq-FW-2`, where the machine has bad rtc.
I did not call `--systohc` so that this system can be used to recreate the issue.

From the log, we can see:
1. In dmesg, rtc is broken:
```
  1 [    2.040000] rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
  2 [    2.040000] rtc-rv8803 11-0032: Voltage low, data loss detected.
  3 [    2.040000] rtc-rv8803 11-0032: An alarm maybe have been missed.
  4 [    2.150000] rtc-rv8803 11-0032: Voltage low, data is invalid.
  5 [    2.150000] rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
```
2. In journal log, systemd thinks it's on a future date `2018-08-18`
```
-- Logs begin at Wed 2017-10-25 22:48:47 UTC, end at Sat 2018-08-18 06:35:17 UTC. --
```

If we get other machines hitting the issue, please help to check what is the output of:
* `hwclock -r`
* `timedatectl`

gkeishin, 2017-11-03:
Sure thanks, that's helpful. I ll get those if I hit again.

gkeishin, 2017-11-03:
```
root@witherspoon:~# uptime
 21:14:18 up 10 min,  load average: 0.49, 1.43, 1.28
root@witherspoon:~# date
Fri Oct 27 21:14:23 UTC 2017
root@witherspoon:~# hwclock -r
Tue Jun  7 19:07:06 2016  0.000000 seconds
root@witherspoon:~# timedatectl
      Local time: Fri 2017-10-27 21:15:57 UTC
  Universal time: Fri 2017-10-27 21:15:57 UTC
        RTC time: Tue 2016-06-07 19:07:16
       Time zone: n/a (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
root@witherspoon:~#
```
Hit again @mine260309 ..

tomjoseph83, 2017-11-03:
This issue would affect journald logging, journal is stopped and metadata written to the journal is lost.

mine260309, 2017-11-06:
@gkeishin On which machine did you hit the issue?

mine260309, 2017-11-06:
On w34 we see the issue now.

The issue probably is not related to time, but related to `machine-id` and some persistent config files.
Some observations:
1. The journal log is now saved at `/var/log/journal/`, on this machine, before deleting the journal logs:
```
root@witherspoon:/tmp# ls -l /var/log/journal/
total 0
drwxr-sr-x    2 root     systemd-       376 Nov  5 04:00 61414e6ffe224cbc8a1a67e1f9a1867e
drwxr-sr-x    2 root     systemd-       376 Nov  5 06:55 7ed39fecdb434d40bff950a0c313c17d
drwxr-sr-x    2 root     systemd-       920 Nov  5 03:56 8d5025ea5e884c3882144ea5cbbe0fae
drwxr-sr-x    2 root     systemd-       376 Nov  5 05:41 c69d08ae6e174ad5b5b11051a126244a
drwxr-sr-x    2 root     systemd-       376 Nov  5 13:30 e05b859ebd694ec1b01f9de14572866f
drwxr-sr-x    2 root     systemd-       376 Nov  5 13:34 f8881623253040c4900690c375b7b375
```
2. Then delete all of them and reboot:
```
root@witherspoon:/var/log/journal# rm -r *
root@witherspoon:/var/log/journal# ls -l
total 0
root@witherspoon:/var/log/journal# reboot
```
3. After reboot, let's see the dir:
```
root@witherspoon:/tmp# ls -l /var/log/journal/
total 0
drwxr-sr-x    2 root     systemd-       376 Nov  5 14:00 e24ad0e2bda24de5b10dfe7f486e12b9
drwxr-sr-x    2 root     systemd-       232 Nov  5 13:56 f8881623253040c4900690c375b7b375
```
While the machine id on it is:
```
root@witherspoon:/tmp# cat /etc/machine-id
8d5025ea5e884c3882144ea5cbbe0fae
```
There are no `8d5025ea5e884c3882144ea5cbbe0fae` dir created at all, and thus we see no logs:
```
root@witherspoon:/tmp# journalctl
No journal files were found.
-- No entries --
```
4. Actually the journalctl is working and the logs are saved on other "machine-id" dirs, we can see the log via `journalctl --merge | tee /tmp/log`.

mine260309, 2017-11-06:
The root cause could be the same as https://unix.stackexchange.com/questions/191313/why-is-my-systemd-journal-not-persistent-across-reboots

swe12345, 2017-11-06:
Observing the same on wsbmc002
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-113-g65edf7d-r10-0-gcdf7635"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-113-g65edf7d-r10"
root@witherspoon:~#
```
```
root@witherspoon:~# uptime
 19:08:06 up  1:54,  load average: 0.18, 0.23, 0.25
root@witherspoon:~# date
Thu Nov  2 19:08:16 UTC 2017
root@witherspoon:~# hwclock -r
Sat May 21 13:54:19 2016  0.000000 seconds
root@witherspoon:~# timedatectl
      Local time: Thu 2017-11-02 19:08:55 UTC
  Universal time: Thu 2017-11-02 19:08:55 UTC
        RTC time: Sat 2016-05-21 13:54:46
       Time zone: n/a (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
root@witherspoon:~#
```

dkodihal, 2017-11-07:
@swe12345 did you notice just the time anomaly, or the journal stopping as well?

dkodihal, 2017-11-07:
We probably need separate issues for the bad RTC and the journal stopping @geissonator and @gkeishin. The problem with the journal may not be related to the RTC problem, as noted by @mine260309. That doesn't explain though why you would not always see the journal issue @mine260309 - I don't for eg see it on certain boxes even after a reboot.

rfrandse, 2017-11-07:
<https://gerrit.openbmc-project.xyz/7822> init: Add init script to setup /etc mount before starting systemd
Resolves  #2542 BMC time is going backwards after a reboot

mine260309, 2017-11-08:
@dkodihal When the issue occurs (e.g. on w34 2 days ago), it is 100% reproducible:
1. Every time BMC reboots, a new dir is created in `/var/log/journal/` with random machine id;
2. The `machine-id` in `/run` and `/etc` are always different, where the one in `/run` is the random one, and the other is the persistent one.

I do not have a chance to check why the issue is not reproduced on some other machines though.

closed 2017-11-28 by geissonator