openbmc/openbmc #2544: [xCAT] upload of  BMC from 1738 to 1740 does not trigger extraction of bmc tar file

cdeadmin, 2017-11-01:


gtmills, 2017-11-01:
This is caused by use deleting /tmp/ every 10 days. This deletes /tmp/images and makes the inotify no longer work.

from /usr/lib/tmpfiles.d/tmp.conf
# Clear tmp directories separately, to make them easier to override
q /tmp 1777 root root 10d

If a node has been up more than 10 days, it isn't going to firmware update without a reboot first.

running  "systemctl restart xyz.openbmc_project.Software.Version.service" on the OpenBMC would also fix the problem.

Need changes to not delete /tmp/images

jayeshmpatel, 2017-11-01:
Do we know why /tmp/images was deleted every 10 days in first place? Do we have any implication with the fix (i.e do not delete every 10 days) like filling up the /tmp after many code update over the many days and not allow the code update in the future?

gtmills, 2017-11-01:
We delete /tmp every 10 days to prevent it from filling up. 
This change will only affect /tmp/images and it should not have any implications. 
The user could fill /tmp/images by uploading many images,  >20, but a user should not have anywhere close to that many images in /tmp/images.

amboar, 2017-11-01:
Please describe the bug in terms of the upstream version tags, not IBM-internal firmware release IDs. I've added the ibm-internal tag, we can remove it with the removal of the internal IDs.

rfrandse, 2017-11-02:
<https://gerrit.openbmc-project.xyz/7731> Exclude /tmp/images from cleanup
Resolves  #2544 Upload of BMC tar file does not trigger extraction after BMC has been up for more than 10 days

closed 2018-01-17 by geissonator