openbmc/openbmc #2545: Journlad corrupted or uncleanly shut down, renaming and replacing

gkeishin, 2017-11-01:
```
root@witherspoon:~# cat /etc/machine-id
a5756e58525c4ea39807aea57192cb93
root@witherspoon:~#
```

`sanpshot`
```
[  OK  ] Reached target Local File Systems.
[   19.850000] systemd[1]: Starting Journal Service...
         Starting Journal Service...
[   20.030000] systemd-journald[955]: File /var/log/journal/a5756e58525c4ea39807aea57192cb93/system.journal corrupted or uncleanly shut down, renaming and replacing.
[  *** ] A start job is running for Journal Service (37s / 1min 30s)[   56.720000] systemd[1]: Started Journal Service.
[  OK  ] Started Journal Service.
         Starting Flush Journal to Persistent Storage...
[   57.930000] systemd-journald[955]: Received request to flush runtime journal from PID 1
[  OK  ] Started Flush Journal to Persistent Storage.
         Starting Create Volatile Files and Directories...

```

````
Oct 25 22:52:10 witherspoon phosphor-bmc-state-manager[1472]: BMC_READY
Oct 25 22:51:38 witherspoon avahi-daemon[976]: Registering new address record for 224.6.6.255 on eth0.IPv4.
Oct 25 22:52:10 witherspoon phosphor-bmc-state-manager[1472]: Setting the BMCState field
Oct 25 22:51:38 witherspoon avahi-daemon[976]: Registering new address record for 127.0.0.1 on eth0.IPv4.
^C
root@witherspoon:~# reboot
[  OK  ] Stopped Phosphor Host Console SSH Per-Connection (9.41.166.76:33684).
[  OK  ] Removed slice system-phosphor\x2dcooling\x2dtype.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Time & Date Service...
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2don.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped Wait for /xyz/openbmc_project/control/host0.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2drunning.slice.
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-obmc\x2dflash\x2dbmc\x2dsetenv.slice.
         Stopping Phosphor Webserver...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped target Host0 running after reset.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-obmc\x2dsync\x2dinventory\x2ditem.slice.
[  OK  ] Removed slice system-dropbear.slice.
[  OK  ] Stopped target Timers.
[  OK  ] Removed slice system-phosphor\x2dmax31785\x2dmsl.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/network/config.
[  OK  ] Removed slice system-phosphor\x2dreset\x2dhost\x2drunning.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/state/chassis0.
[  OK  ] Removed slice system-phosphor\x2ddiscover\x2dsystem\x2dstate.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Chassis0 power on after reset.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped Assert bmc_booted LED.
[  OK  ] Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
[  OK  ] Stopped target Phosphor System - Standby.
         Stopping Phosphor Version Software Manager...
         Stopping Phosphor User Manager...
         Stopping Phosphor DBus Monitor...
[... 520 lines cut ...]
[  OK  ] Found device /dev/ttyS4.
[  OK  ] Found device /dev/ttyVUART0.
[  OK  ] Created slice system-xyz.openbmc_project.Hwmon.slice.
         Starting Prepare for overlay fs...
         Mounting /var/volatile...
[  OK  ] Started Prepare for overlay fs.
[  OK  ] Mounted /var/volatile.
[  OK  ] Created slice system-xyz.openbmc_project.led.controller.slice.
         Starting Prepare for overlay fs...
         Starting Load/Save Random Seed...
[  OK  ] Started Prepare for overlay fs.
[  OK  ] Started Load/Save Random Seed.
         Mounting /home...
         Mounting /etc...
[  OK  ] Mounted /etc.
[  OK  ] Mounted /home.
[  OK  ] Reached target Local File Systems.
[   19.850000] systemd[1]: Starting Journal Service...
         Starting Journal Service...
[   20.030000] systemd-journald[955]: File /var/log/journal/a5756e58525c4ea39807aea57192cb93/system.journal corrupted or uncleanly shut down, renaming and replacing.
[  *** ] A start job is running for Journal Service (37s / 1min 30s)[   56.720000] systemd[1]: Started Journal Service.
[  OK  ] Started Journal Service.
         Starting Flush Journal to Persistent Storage...
[   57.930000] systemd-journald[955]: Received request to flush runtime journal from PID 1
```

geissonator, 2017-11-02:
When I looked at this specific system the /var/ filesystem was full.  The main users were the debug tar file and /var/log/wtmp.  We cleared both out and system got back to working.  I have #2488 to fix the wtmp issue and sounds like test team is/has moved debug to /tmp/.  Let me know if you see this again George in a situation where persistent filesystem is not full.

geissonator, 2017-11-09:
Another issue was systemd and it's use of /etc/machine-id.  Eddie has a fix for that up here - https://gerrit.openbmc-project.xyz/#/c/7822/

geissonator, 2017-12-14:
I think we're good on this one now.

closed 2017-12-14 by geissonator

Rtoax, 2022-06-07:
i got `File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.` ERROR in `systemd-journal-remote.service`

```
Jun 07 10:57:49 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:49 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:50 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:50 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:51 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:51 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:53 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:54 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:57 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:57 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
Jun 07 10:57:58 hg-68 systemd-journal-remote[17364]: File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.
```

and each journal remote log`remote-10.253.32.164.journal` renamed:

```
   539  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c522c8ce-bb3745d144ef9bb5.journal~
   540  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c529fddb-e0bc4bdf51cddc1a.journal~
   541  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c52ded99-4806e98182050b1f.journal~
   542  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c5320b98-909bf7d5b0308a51.journal~
   543  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c53670c9-2b384082ab81ef89.journal~
   544  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:58 remote-10.253.32.164@0005e0d2c53d7d61-49f6c709ee1b56f6.journal~
   545  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c541c993-90f5258c180f5761.journal~
   546  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c545a3cb-36f73bff605920cd.journal~
   547  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c54a03f6-78b8d9b2f24a0627.journal~
   548  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c54e6cc6-d6afc06b7b24c065.journal~
   549  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c55261c3-3f26e7ff911b2f36.journal~
   550  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164@0005e0d2c558314b-cd30f788c13467b0.journal~
   551  -rw-r----- 1 systemd-journal-remote systemd-journal-remote 8388608 Jun  7 10:59 remote-10.253.32.164.journal
```

williamspatrick, 2022-06-07:
> i got `File /home/journal-remote/remote-10.253.32.164.journal corrupted or uncleanly shut down, renaming and replacing.` ERROR in `systemd-journal-remote.service`
>
> ```
> Jun 07 10:57:49 hg-68 [...]

This appears to be an unrelated issue. I suggest opening another issue if appropriate. 

Why do you have systemd-journal-remote enabled?  That is not something done by our default installation.

Rtoax, 2022-06-07:
@williamspatrick Yep, you're right, https://github.com/systemd/systemd/issues/23651