openbmc/openbmc #2554: P9 OCC control objects are not created after BMC update while host running

mine260309, 2017-11-02:
On P9 machines, OCC control objects are not created with below steps:
1. Host is running
2. Update BMC
3. After BMC reboot, occ control objects are not created, while the host is still running.
They won't be created unless host is rebooted.

Root cause:
* The P9 OCC control objects are created when cpu inventory are created;
* After BMC update, the inventory is cleared;
* After BMC reboot, there is not inventory information, and thus the occ control objects are not created;

Should the inventory information be preserved on BMC update?

geissonator, 2017-11-02:
We don't officially support BMC updates while the system is running.  This is def a design point we'll need to work in when we do.

geissonator, 2017-11-02:
FYI @anoo1 and @dkodihal

dkodihal, 2017-11-02:
I suppose the update shouldn't be clearing the app specific persisted data under /var/lib/* @anoo1?

dkodihal, 2017-11-02:
Probably similar to #2532.

anoo1, 2017-11-02:
If we can get a list of the REST objects on the /org/open_power/control done in between the steps to narrow down when are the objects being removed or not recreated.

closed 2019-04-26 by stale[bot]