openbmc/openbmc #2555: [xCAT] Request support to generate dumps in a way that can scale...

whowutwut, 2017-11-02:
I'm trying to create some function support in xCAT to list/generate/delete Dumps on the BMC, described here: https://github.com/xcat2/xcat-core/issues/4235

In writing this issue, I'm also trying to think about how to cover some use cases at scale.   This may or may not already exist, so just seeing if someone can confirm for me... 

What I'm trying to do is... We find a problem on X number of nodes.  And I want to trigger a dump across all nodes using xCAT and then retrieve the dumps to send over to the support teams.  What I would do is the following: 
1. Across all the BMCs, trigger a new dump generating
2. Wait however long.... 
3. For each BMC, list out the dumps, and find the last one, and download that one ...
4. Before going on, rename the obmcdump_<ID>_elapsed.tar.xz file to something else... 
This does not scale well....

I'd like to push this over to the BMC because I much like to avoid downloading the file to a temporary location for EACH BMC, renaming the file to something better, then copying it over to the final location, etc etc,....  So asking if there's some feature that we can request that could be added in fast path.... 
* If we provide -1 as the dump ID, we download the "latest" dump that was generated
* In the CreateDump command, can we pass in additional data to specify the name of the dump file we want it to have? or appended to the generic name?   If we want to download 1000 dumps to 1 place, how can we tell which BMC it belongs to?
* Is there a way to tell that a dump is in generation?  we can block until the dump is complete, then retrieve it...   and this ties into the first request...  asking for the last one created.

dkodihal, 2017-11-03:
When a dump is created, the dump id is returned to you - https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Dump/Create.interface.yaml. That's how you know which dump to download. @ojayanth you should make this obvious in https://gerrit.openbmc-project.xyz/#/c/7066/1/README.md.

whowutwut, 2017-11-03:
Yes, @dkodihal Thanks, I do see this now... 

```
[root@mgt03 p9]# ./create_dump.sh 
++ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -d '{"data": []}' -X POST https://192.168.74.172/xyz/openbmc_project/dump/action/CreateDump
{
  "data": 4, 
  "message": "200 OK", 
  "status": "ok"
}
```

closed 2019-04-27 by stale[bot]