openbmc/openbmc #2556: [xCAT] Got "Internal Server Error" when get network information

xuweibj, 2017-11-03:
Version:
ibm-v1.99.10-113-g65edf7d-r3-0-g9e4f715
IBM-witherspoon-ibm-OP9_v1.19_1.68

Test script: (test.sh)
```
curl -k -c cjar -H "Content-Type: application/json" -d '{ "data": ["root", "0penBmc"] }' https://172.12.253.107/login
curl -k -b cjar -X POST -H "Content-Type: application/json" -d '{"data":["eth0","11"]}' https://172.12.253.107/xyz/openbmc_project/network/action/VLAN
curl -k -b cjar -X POST -H "Content-Type: application/json" -d '{"data":["xyz.openbmc_project.Network.IP.Protocol.IPv4","172.11.139.11",16,"0.0.0.0"]}' https://172.12.253.107/xyz/openbmc_project/network/eth0_11/action/IP
curl -k -b cjar -X GET -H "Content-Type: application/json" https://172.12.253.107/xyz/openbmc_project/network/enumerate
```

Result:
```
#sh test.sh
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException(\"Unknown interface 'xyz.openbmc_project.Object.Delete'.\",)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 798, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 855, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1038, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1012, in wrap",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 977, in wrap",
      "    data = callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 199, in __call__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 305, in do_get",
      "    mapper_data=request.route_data['map']).dataitems()}",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 187, in enumerate_subtree",
      "    path, owner, interfaces, match))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 110, in __get_properties_on_bus",
      "    properties_iface, i))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 96, in __get_properties_on_iface",
      "    properties_iface.GetAll, iface)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 82, in __try_properties_interface",
      "    return f(*a)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'xyz.openbmc_project.Object.Delete'."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}
```

When I run every curl command manually, I never got this response. So I guess that we need to wait several seconds after adding new IP object. But could this be better?

@whowutwut @zet809

I do not know whether this is a bug, so have not create an issue in CQ.

whowutwut, 2017-11-03:
@xuweibj Please also put in the test machine you ran this against...  so that I could follow up...

whowutwut, 2017-11-03:
Ok, found the node: mid05tor12cn05 

```
Nov  3 02:51:13 briggs01 dhcpd: DHCPOFFER on 172.12.253.107 to 70:e2:84:14:29:0f (witherspoon) via enP34p1s0f0.12
Nov  3 02:51:13 briggs01 dhcpd: DHCPREQUEST for 172.12.253.107 (172.12.253.27) from 70:e2:84:14:29:0f (witherspoon) via enP34p1s0f0.12
Nov  3 02:51:13 briggs01 dhcpd: DHCPACK on 172.12.253.107 to 70:e2:84:14:29:0f (witherspoon) via enP34p1s0f0.12
```

BMC mac is 1 hex under the eth0
```
[root@briggs01 p9]# lsdef mid05tor12cn05 -i mac
Object name: mid05tor12cn05
    mac=70:e2:84:14:28:0e
```

Could running those above commands have "hosed" the BMC or put the networking into  a bad state where BMC Reset does not clean up correctly and resulted in:  https://w3.rchland.ibm.com/projects/bestquest/?defect=SW407037 ?

xuweibj, 2017-11-06:
Hi @whowutwut , the node I used is mid05tor12cn11, not mid05tor12cn05.

amboar, 2017-11-13:
@xuweibj @whowutwut this is a public issue tracker. Please describe the issues in terms of changes, tags and machines that are also publicly accessible. If any of those items cannot be reached publicly then the issue should not be filed here. Please do not reference or link to IBM internal bugs, these are also unreachable by non-IBM people participating in the project.

gtmills, 2018-10-12:
@xuweibj @whowutwut Still a problem or can we close? Also, this refers to an IBM internal release, if this is still a problem please update with a non-internal release.

gtmills, 2019-01-23:
No response. Closing.

closed 2019-01-23 by gtmills