openbmc/openbmc #2557: Failed to start Factory reset BMC if reset enabled.

geissonator, 2017-11-03:
Saw this in a CI run.  A PDU cycle fixed the issue so I don't think it was related to the changes being tested (they looked pretty benign). 

Here's the serial:
```
[  OK  ] Started Apply Kernel Variables.
         Starting Create Static Device Nodes in /dev...
[  OK  ] Started Create Static Device Nodes in /dev.
         Starting udev Kernel Device Manager...
[  OK  ] Reached target Local File Systems (Pre).
[   14.870000] systemd[1]: Mounting /media...
         Mounting /media...
[   14.870000] systemd[1]: Mounting /usr/local...
         Mounting /usr/local...
[   15.100000] systemd[1]: Started udev Kernel Device Manager.
[  OK  ] Started udev Kernel Device Manager.
[   15.340000] systemd[1]: Mounted /media.
[  OK  ] Mounted /media.
[   15.590000] systemd[1]: Started udev Coldplug all Devices.
[  OK  ] Started udev Coldplug all Devices.
[   15.850000] systemd[1]: Mounted /usr/local.
[  OK  ] Mounted /usr/local.
[   16.400000] systemd[1]: Found device /dev/aspeed-lpc-ctrl.
[  OK  ] Found device /dev/aspeed-lpc-ctrl.
[   16.690000] systemd[1]: Found device /dev/aspeed-mbox.
[  OK  ] Found device /dev/aspeed-mbox.
[   16.990000] systemd[1]: Started Factory reset BMC if reset enabled.
[  OK  ] Started Factory reset BMC if reset enabled.
[   16.990000] systemd[1]: Found device /dev/ttyS4.
[  OK  ] Found device /dev/ttyS4.
[  OK  ] Found device /dev/ttyVUART0.
[  OK  ] Created slice system-xyz.openbmc_project.Hwmon.slice.
         Starting Factory reset BMC if reset enabled...
[  OK  ] Started Factory reset BMC if reset enabled.
         Starting Factory reset BMC if reset enabled...
[  OK  ] Started Factory reset BMC if reset enabled.
[FAILED] Failed to start Factory reset BMC if reset enabled.
See 'systemctl status obmc-flash-bmc-reset.service' for details.
[DEPEND] Dependency failed for /var.
[DEPEND] Dependency failed for Network Time Synchronization.
[DEPEND] Dependency failed for Load/Save Random Seed.
[DEPEND] Dependency failed for Avahi mDNS/DNS-SD Stack Activation Socket.
[DEPEND] Dependency failed for Avahi mDNS/DNS-SD Stack.
[DEPEND] Dependency failed for Flush Journal to Persistent Storage.
[DEPEND] Dependency failed for Virtual Machine and Container Storage.
[DEPEND] Dependency failed for Bind mount volatile /var/lib.
[DEPEND] Dependency failed for Basic System.
[DEPEND] Dependency failed for Multi-User System.
[DEPEND] Dependency failed for Phosphor System - Standby.
[DEPEND] Dependency failed for Containers.
[DEPEND] Dependency failed for Prepare for overlay fs.
[DEPEND] Dependency failed for /etc.
[DEPEND] Dependency failed for Local File Systems.
[DEPEND] Dependency failed for Mount machine-id.
[DEPEND] Dependency failed for /home.
[DEPEND] Dependency failed for Emergency Shell.
[DEPEND] Dependency failed for Emergency Mode.
[DEPEND] Dependency failed for /var/volatile.
[  OK  ] Reached target Timers.
[FAILED] Failed to start Factory reset BMC if reset enabled.
See 'systemctl status obmc-flash-bmc-reset.service' for details.
[   19.910000] systemd[1]: Dependency failed for /var.
[DEPEND] Dependency failed for /var.
[   19.940000] systemd[1]: Dependency failed for Prepare for overlay fs.
[DEPEND] Dependency failed for Prepare for overlay fs.
```

Here's some data from the previous boot, it was doing a code update to a new image:
```
Nov 03 12:27:44 witherspoon kernel: block ubiblock0_0: released
Nov 03 12:27:44 witherspoon kernel[981]: [10427.990000] block ubiblock0_0: released
[... 401 lines cut ...]
Nov 03 12:30:36 witherspoon systemd[1]: Stopped target Swap.
Nov 03 12:30:36 witherspoon systemd[1]: Unmounting /var...
Nov 03 12:30:36 witherspoon systemd[1]: Unmounting /usr/local...
Nov 03 12:30:36 witherspoon systemd[1]: Unmounting /media...
Nov 03 12:30:36 witherspoon systemd[1]: etc.mount: Mount process exited, code=exited status=1
Nov 03 12:30:36 witherspoon systemd[1]: Failed unmounting /etc.
Nov 03 12:30:36 witherspoon systemd[1]: Unmounted /usr/local.
Nov 03 12:30:36 witherspoon systemd[1]: Unmounted /media.
Nov 03 12:30:36 witherspoon systemd[1]: var.mount: Mount process exited, code=exited status=1
Nov 03 12:30:36 witherspoon systemd[1]: Failed unmounting /var.
Nov 03 12:30:36 witherspoon systemd[1]: Reached target Unmount All Filesystems.
Nov 03 12:30:37 witherspoon systemd[1]: Stopped target Local File Systems (Pre).
Nov 03 12:30:37 witherspoon systemd[1]: Stopped Remount Root and Kernel File Systems.
Nov 03 12:30:37 witherspoon systemd[1]: Stopped Create Static Device Nodes in /dev.
Nov 03 12:30:37 witherspoon systemd[1]: Reached target Shutdown.
Nov 03 12:30:37 witherspoon systemd[1]: Reached target Final Step.
Nov 03 12:30:37 witherspoon systemd[1]: Starting Reboot...
Nov 03 12:30:37 witherspoon systemd[1]: Shutting down.
Nov 03 12:30:38 witherspoon systemd[1]: Hardware watchdog 'aspeed_wdt', version 0
Nov 03 12:30:38 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Nov 03 12:30:38 witherspoon kernel: systemd-shutdow: 66 output lines suppressed due to ratelimiting
Nov 03 12:30:38 witherspoon systemd[1]: Set hardware watchdog to 10min.
Nov 03 12:30:39 witherspoon systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Nov 03 12:30:39 witherspoon systemd-journald[957]: Journal stopped
```

ojayanth, 2017-11-12:
me and @gkeishin was trying to validate the factory reset after code update( R5->R8) . Looks like we hit similar issue “Failed to start Factory reset BMC if reset enabled. #2557”. After that we did same test , R5 to R8 upgrade and tried factory reset steps skipping “FieldModeEnabled” enabled one . In this case system worked good after pdu reset. Based on this experiment looks like some potential file corruption in the “FieldModeEnabled” path.

geissonator, 2018-01-03:
Haven't seen this one in a while.  I assume moving the factory reset code into uboot helped.

closed 2018-01-03 by geissonator