openbmc/openbmc #2559: Core dumping process just keeps on restarting

geissonator, 2017-11-03:
Not sure what's up, but our service restart limits is not working here:

```
Nov 03 15:52:15 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Nov 03 15:52:15 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Nov 03 15:52:15 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:15 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Nov 03 15:52:15 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:15 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/59
Nov 03 15:52:16 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Nov 03 15:52:16 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Nov 03 15:52:16 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Nov 03 15:52:16 witherspoon openpower-occ-control[5782]: Failed to get property
Nov 03 15:52:16 witherspoon openpower-occ-control[5782]: The operation failed internally.
Nov 03 15:52:16 witherspoon openpower-occ-control[5782]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Nov 03 15:52:16 witherspoon openpower-occ-control[5782]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Nov 03 15:52:16 witherspoon systemd[1]: Started Process Core Dump (PID 5788/UID 0).
Nov 03 15:52:17 witherspoon systemd-coredump[5425]: Process 5415 (openpower-occ-c) of user 0 dumped core.
Nov 03 15:52:19 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:17 UTC 2017 Successfully completed
Nov 03 15:52:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Nov 03 15:52:20 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:18 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/60
Nov 03 15:52:19 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Nov 03 15:52:20 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:18 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/61
Nov 03 15:52:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:21 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:19 UTC 2017 Successfully completed
Nov 03 15:52:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Nov 03 15:52:21 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:19 UTC 2017 Successfully completed
Nov 03 15:52:19 witherspoon systemd-coredump[5371]: Removed old coredump core.openpower-occ-c.0.3fc21f50d38040a4ad79116aeb07af05.5415.1509724303000000000000.xz.
Nov 03 15:52:19 witherspoon systemd-coredump[5371]: Process 5369 (openpower-occ-c) of user 0 dumped core.
Nov 03 15:52:20 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Nov 03 15:52:20 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Nov 03 15:52:20 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Nov 03 15:52:22 witherspoon openpower-occ-control[5838]: Failed to get property
Nov 03 15:52:22 witherspoon openpower-occ-control[5838]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Nov 03 15:52:22 witherspoon openpower-occ-control[5838]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Nov 03 15:52:22 witherspoon openpower-occ-control[5838]: The operation failed internally.
Nov 03 15:52:22 witherspoon systemd[1]: Started Process Core Dump (PID 5859/UID 0).
Nov 03 15:52:23 witherspoon phosphor-dump-manager[1036]: cp: can't stat '/var/lib/systemd/coredump/core.openpower-occ-c.0.3fc21f50d38040a4ad79116aeb07af05.5415.1509724303000000000000.xz': No such file or directory
Nov 03 15:52:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Nov 03 15:52:23 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Nov 03 15:52:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Nov 03 15:52:25 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Nov 03 15:52:25 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Nov 03 15:52:25 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Nov 03 15:52:25 witherspoon openpower-occ-control[5927]: Failed to get property
Nov 03 15:52:25 witherspoon openpower-occ-control[5927]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Nov 03 15:52:25 witherspoon openpower-occ-control[5927]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Nov 03 15:52:25 witherspoon openpower-occ-control[5927]: The operation failed internally.
Nov 03 15:52:26 witherspoon systemd[1]: Started Process Core Dump (PID 5935/UID 0).
Nov 03 15:52:28 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:28 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/62
Nov 03 15:52:29 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:29 UTC 2017 Successfully completed
Nov 03 15:52:29 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:29 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/63
Nov 03 15:52:30 witherspoon phosphor-dump-manager[1036]: Fri Nov 3 15:52:30 UTC 2017 Successfully completed
Nov 03 15:52:31 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Nov 03 15:52:31 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Nov 03 15:52:31 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:31 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Nov 03 15:52:33 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Nov 03 15:52:33 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Nov 03 15:52:33 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Nov 03 15:52:33 witherspoon openpower-occ-control[5985]: Failed to get property
Nov 03 15:52:33 witherspoon openpower-occ-control[5985]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Nov 03 15:52:33 witherspoon openpower-occ-control[5985]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Nov 03 15:52:33 witherspoon openpower-occ-control[5985]: The operation failed internally.
Nov 03 15:52:33 witherspoon systemd[1]: Started Process Core Dump (PID 5986/UID 0).
Nov 03 15:52:35 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Nov 03 15:52:35 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Nov 03 15:52:35 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:35 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Nov 03 15:52:36 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
[... 32 lines cut ...]
8	./dumps/49
8	./dumps/50
8	./dumps/51
8	./dumps/52
8	./dumps/53
44	./dumps/54
44	./dumps/55
8	./dumps/56
44	./dumps/57
44	./dumps/58
8	./dumps/59
44	./dumps/60
44	./dumps/61
8	./dumps/62
44	./dumps/63
44	./dumps/64
8	./dumps/65
44	./dumps/66
8	./dumps/67
44	./dumps/68
44	./dumps/69
44	./dumps/70
808	./dumps
808	.
```

geissonator, 2017-11-03:
It looks like the inventory service had hit a failure, so I guess that's why the get property was failing.

ojayanth, 2017-11-06:
@geissonator Please share the /var/lib/phosphor-debug-collector/dumps directory size info. 
Is it exceeds 1MB?

geissonator, 2018-01-03:
The initial code change for this is out in https://gerrit.openbmc-project.xyz/#/c/6540/

When you look at the above traces you can see we're outside the 10s window for the retry threshold:
```
Nov 03 15:52:15 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:15 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump’.
--> Restart 1
Nov 03 15:52:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump’.
--> Restart 2
Nov 03 15:52:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump’.
--> Restart 3
Nov 03 15:52:31 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Nov 03 15:52:31 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump
--> Restart 4 (exceeds the 3) but we're past the 10s burst limit so our count is restarted 
```

I think the issue here is how long the core dump process is taking.  That's disabled in the current code due to filesystem corruption issues.  I verified with our current code in master that we properly stop after 3 restarts within 10s
```
Jan 03 19:16:42 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Failed with result 'exit-code'.
Jan 03 19:16:44 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Service hold-off time over, scheduling restart.

Jan 03 19:16:44 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Failed with result 'exit-code'.
Jan 03 19:16:45 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Service hold-off time over, scheduling restart.

Jan 03 19:16:45 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Failed with result 'exit-code'.
Jan 03 19:16:46 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Service hold-off time over, scheduling restart.

Jan 03 19:16:46 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Start request repeated too quickly.
```

rfrandse, 2018-01-03:
<https://gerrit.openbmc-project.xyz/8400> Reduce service restart limit from 3 to 2 on fails
Resolves  #2559 Core dumping process just keeps on restarting

closed 2018-01-23 by geissonator