openbmc/openbmc #2560: Enhance thermal shutdown error metadata and logging

geissonator, 2017-11-03:
We had a system hit a thermal issue, the appropriate errors were logged but it would be good to add a few things.

- Add the temperature threshold and the read temperature to the metadata
```
 |------------------------------------------------------------------------------|
 |  Id      Date Time           Message                         Severity        |
 |------------------------------------------------------------------------------|
 |       6  11/02/2017 13:32:15 xyz.openbmc_project.State.Shutdown.ThermalEvent.Error.ProcessorError           |
 |------------------------------------------------------------------------------|
 |Additional Data : _PID=1136                                                   |

 |------------------------------------------------------------------------------|
 |  Id      Date Time           Message                         Severity        |
 |------------------------------------------------------------------------------|
 |       7  11/02/2017 13:32:16 xyz.openbmc_project.Sensor.Threshold.Error.CriticalHighError           |
 |------------------------------------------------------------------------------|
 |Additional Data : SENSOR_TYPE=temperature                                     |
 |Additional Data : _PID=1136       
```
- Reduce the amount of logging on the fail, we got one for every core
```
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
Nov 02 18:31:52 witherspoon phosphor-dbus-monitor[1136]: The system is too hot.  Shutting down.
```

geissonator, 2017-11-03:
FYI @msbarth @spinler

rfrandse, 2018-02-20:
<https://gerrit.openbmc-project.xyz/9073> Witherspoon processor overtemp path changes
Resolves  #2560 Enhance thermal shutdown error metadata and logging

closed 2018-02-28 by geissonator