openbmc/openbmc #2562: Link Local IPv4 object doesnt reinstate post IP Delete and Add

gkeishin, 2017-11-06:
- [x] Delete the IP 
- [x] Add the IP back 
- [x] Link Local object never creates back

`NW object post adding back`
```
/xyz/openbmc_project/network/eth0/enumerate
{
    "/xyz/openbmc_project/network/eth0/ipv4/5629cdfc": {
        "Address": "9.3.21.42", 
        "Gateway": "9.3.20.1", 
        "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static", 
        "PrefixLength": 23, 
        "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    }, 
    "/xyz/openbmc_project/network/eth0/ipv6/913ae752": {
        "Address": "fe80::72e2:84ff:fe14:2876", 
        "Gateway": "", 
        "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.LinkLocal", 
        "PrefixLength": 64, 
        "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv6"
    }
}
```

`ip addr shows though`
```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet 169.254.117.63/16 brd 169.254.255.255 scope link eth0
       valid_lft forever preferred_lft forever
    inet xx.xx.xx.xx/23 brd 9.3.21.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link 
       valid_lft forever preferred_lft forever
root@witherspoon:~# 
```

gkeishin, 2017-11-06:
`Driver Info`
```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-67fda6b3a8b85e3063fc46fb3262ffb130ba396d #1 Wed Nov 1 07:50:18 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-3"
VERSION_ID="v2.0-3-g27b199a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-3"
BUILD_ID="v2.0"
root@witherspoon:~# uptime
 23:57:05 up 39 min,  load average: 0.48, 0.43, 0.48
root@witherspoon:~#
```

- [x] Journald 
[Network_core_dump.txt](https://github.com/openbmc/openbmc/files/1447028/Network_core_dump.txt)

rfrandse, 2017-11-09:
<https://gerrit.openbmc-project.xyz/7832> Don't throw exception in the case of deletion of link local address.
Resolves  #2562 Link Local IPv4 object doesnt reinstate post IP Delete and Add

rfrandse, 2017-11-15:
<https://gerrit.openbmc-project.xyz/7863> Create netlink socket outside from rtnetlink server
Resolves  #2562 Link Local IPv4 object doesnt reinstate post IP Delete and Add

closed 2017-11-16 by geissonator