openbmc/openbmc #2563: OP910:OPRA:witherspoon:no fault led on when PSU was removed from the system

cdeadmin, 2017-11-06:


bjwyman, 2017-11-06:
On a Witherspoon system with VERSION_ID="ibm-v1.99.10-113-g65edf7d-r1-0-g84427f0", a power supply was removed from the system. A check of inventory showed:
```
"/xyz/openbmc_project/inventory/system/chassis/power_supply0": {
 "Present": 1,
 "PrettyName": "power_supply0"
},
"/xyz/openbmc_project/inventory/system/chassis/power_supply1": {
 "Present": 0,
 "PrettyName": "power_supply1"
},
```
So, power_supply1 was seen as not present. However, the expectation was that a fault LED would turn on (enclosure fault at least). No fault LED turned on.

REST command showing LED off:
```
$ restcmd -H wsp005 -u   /xyz/openbmc_project/led/groups/system_fault/attr/Asserted -o GET
 "data": 0,
```

bjwyman, 2017-11-06:
# **STATUS**
**11/06**
- Today: Got this "bridged" over from internal defect. I have some commits up in Gerrit that already address the main issue already known. Discussed possible lack of error if remove when powered on already, but not certain if hitting some mount overlay issue during testing.
- Tomorrow: Re-test various scenarios of power supply missing.
- Blockers: None.

bjwyman, 2017-11-06:
# Task List
- [ ] Update strings for GPIO presence monitor.
- [ ] Update strings for PSU fault monitor.
- [ ] Update strings for power supply not present errors.

amboar, 2017-11-13:
@bjwyman Please reference only publicly accessible versions and tags, and describe publicly accessible changes in issues filed in the public bug tracker. Referencing IBM-internal issues and tags is unhelpful to others in the community.

bjwyman, 2017-11-13:
I am assuming the bad reference here is the internal IBM build VERSION_ID.

This probably applies to pretty much all OpenBMC builds so far, but the most recent external tag that this would apply to is v1.99.10.

amboar, 2017-11-13:
Also to the summary line, which references internal firmware release bundles

rfrandse, 2017-11-15:
<https://gerrit.openbmc-project.xyz/7878> Fix power supply presence information
Resolves:  #2563 witherspoon:no fault led on when PSU was removed from the system

closed 2017-11-29 by geissonator