openbmc/openbmc #2564: After activating debug image, on multiple reboots switched back to old image

prkatti1, 2017-11-07:
Pre-condition:
System is in dhcp mode
 version:

ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-113-g65edf7d-r9-0-g67617be"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-113-g65edf7d-r9"

Steps:
1. updated and activated debug image
2. on 1st reboot, sytstem was showing debug image version
3. After multiple reboots it switched back to old build

{
  "data": {
    "/xyz/openbmc_project/software/68967d96": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active",
      "Version": "v1.99.10-225-gd662943",
      "associations": [
        [
          "inventory",
          "activation",
          "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
      ]
    },
    "/xyz/openbmc_project/software/68967d96/inventory": {
      "endpoints": [
        "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
      ]
    },
    "/xyz/openbmc_project/software/68967d96/software_version": {
      "endpoints": [
        "/xyz/openbmc_project/software"
      ]
    },
    "/xyz/openbmc_project/software/776f6a74": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.7-1421-g2f4cbfe,buildroot-2017.08-6-g319c6e1,skiboot-v5.8-130-g712837cedca0,hostboot-8e7e2ad,linux-4.13.4-openpower1-p1c668f0,petitboot-v1.6.0-p85a1eab,machine-xml-1bd247d-p7c6c471,occ-60c09cd,hostboot-binaries-b7d8417,capp-ucode-p9-dd2-v1,sbe-ff713df",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-ibm-OP9_v1.19_1.52",
      "associations": [
        [
          "inventory",
          "activation",
          "/xyz/openbmc_project/inventory/system/chassis"
        ]
      ]
    },
    "/xyz/openbmc_project/software/776f6a74/inventory": {
      "endpoints": [
        "/xyz/openbmc_project/inventory/system/chassis"
      ]
    },
    "/xyz/openbmc_project/software/776f6a74/software_version": {
      "endpoints": [
        "/xyz/openbmc_project/software"
      ]
    },
    "/xyz/openbmc_project/software/active": {
      "endpoints": [
        "/xyz/openbmc_project/software/776f6a74",
        "/xyz/openbmc_project/software/e0e1ac80",
        "/xyz/openbmc_project/software/68967d96"
      ]
    },
    "/xyz/openbmc_project/software/e0e1ac80": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Priority": 1,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "ibm-v1.99.10-113-g65edf7d-r9-0-g67617be",
      "associations": [
        [
          "inventory",
          "activation",
          "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
      ]
    },
    "/xyz/openbmc_project/software/e0e1ac80/inventory": {
      "endpoints": [
        "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
      ]
    },
    "/xyz/openbmc_project/software/e0e1ac80/software_version": {
      "endpoints": [
        "/xyz/openbmc_project/software"
      ]
    },
    "/xyz/openbmc_project/software/functional": {
      "endpoints": [
        "/xyz/openbmc_project/software/776f6a74",
        "/xyz/openbmc_project/software/e0e1ac80"
      ]
    }
  },
  "message": "200 OK",
  "status": "ok"

amboar, 2017-11-13:
@prkatti1 This issue references an IBM-internal tag. Please edit the issue to describe it in terms of publicly accessible tags. If that can't be done, then it's likely this issue should not be in the public issue tracker.

prkatti1, 2017-11-16:
closing this as discussed with George, known issue.

closed 2017-11-16 by prkatti1