openbmc/openbmc #2575: During link local address delete network daemon restarted

cdeadmin, 2017-11-10:


amboar, 2017-11-13:
This issue has been tagged as `ibm-internal` because it references IBM-internal tags, commits, machines or issues. This is a public issue tracker and is for the benefit of the community. If any of the details are not publicly accessible then the issue should be filed elsewhere, or reworded to reflect the relevant publicly accessible details.

rfrandse, 2017-11-15:
network configuration will not work consistently via REST interface. It should work with in band - ipmitool interface. This will impact network configuration functions.

rfrandse, 2017-11-15:
<https://gerrit.openbmc-project.xyz/7863> Create netlink socket outside from rtnetlink server
Resolves  #2575 During link local address delete network daemon restarted

closed 2017-11-16 by geissonator