openbmc/openbmc #2579: Sprint 17.11.29 - Committed

zahrens, 2017-11-11:


dkodihal, 2017-11-30:
***Squad 1 sprint plan, WIP***

**Jayanth**
1) Wrap up #2359 - dreport tool adoption for collecting debug data.
2) SW409162 : OP910: 500 Internal Server Error - DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.164 was not provided by any .service files"
3) #2690 OpenBMC Dump implementation : add unit tests
4) #2689 Document OpenBMC Dump API and usage

**Devender**
1) #2551 Add inventory UT
2) Close #2653 BMC went into bad state, not responding to REST
3) #2691 Phosphor-logging : document API and usage

**Aditya**
1) #2550 Add host ipmi UT
2) #2507 IPMID logging footprint 

**Naveen**
1) #2549 Add errorlog UT

**Ratan**
1) Wrap/close #2450 Setting hostname does not persist in the BMC - Removing network files and reboot causes this
2) Wrap up #2254 Implement event logging mechanism.
3) Wrap up #2399 Yaml changes to create events.
4) #2547  Create event for network changes
5) SW404936 : [xCAT] Need a way to modify a single attribute for a existing Network object via REST
6) #2692 Phosphor-networkd : document API

**Nagaraju**
1) Wrap/close #1496 ObjectMapper [Feature Request]: Get service "well-known" names.
2) Wrap up #2670 [System] : User initiated dump request throwing "501 Not Implemented" error after dump service restart.
3) #2546 Network unit tests 
4) SW407884 : OpenBMC:Witherspoon:FVT- MAC address from REST will not show 0 when first nibble is 0
5) SW409493 : STC:OP910:Witherspoon: BMC Error Log, logs 100 errors and quits logging
6) SW407820 : Don't throw exception while validating the IP configuration.
7) SW405141 : Witherspoon:FVT:910- BMC allows to configure invalid gateway address
8) More?

**Dhruv**
1) SW409770 - Fixup PowerSupplyRedundancy object path 
2) Wrap up #2319 Events : persist using Cereal.
3) #2347 Failed to start SYNC system BMC ethernet.
4) #2154 Create power supply redundancy and derating factor sensors

**Tom**
1) Close #2656 Mapper Error- Error starting event loop: Connection timed out
2) #2693  Design a mapping from MRW to IPMI SDR 
3) Pick defects and/or ipmid logging footprint?

**Deepak**
1) SW408866 : [Witherspoon] : Event notification stops working when events are filtered with only interfaces (inconsistent)
2) #2694 phoshor-settings : document usage 
3) Error logging cap proposal?

ngorugan, 2017-12-14:
**Status**
  Nagaraju

1. Wrap/close #1496 ObjectMapper [Feature Request]: Get service "well-known" names(Brad has to merge the code).
2. Wrap up #2670 [System] : User initiated dump request throwing "501 Not Implemented" error after dump service restart(Made changes on top of Brad's changes for #1496)
3.  #2546 Network unit tests (Yet to start, understanding gtest and gmock testsuite framework)
4.     SW407884 : OpenBMC:Witherspoon:FVT- MAC address from REST will not show 0 when first nibble is 0(code got merged into master)
5.     SW409493 : STC:OP910:Witherspoon: BMC Error Log, logs 100 errors and quits logging(code got merged into master)
6.     SW407820 : Don't throw exception while validating the IP configuration.(got under review)
7.     SW405141 : Witherspoon:FVT:910- BMC allows to configure invalid gateway address(code got merged into master)

closed 2018-01-02 by rfrandse