openbmc/openbmc #2591: openpower-update-manager  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally. - OP910 1742E

cdeadmin, 2017-11-11:


anoo1, 2017-11-11:
```
Nov 10 16:27:21 witherspoon openpower-update-manager[1142]: Error in Mapper call
Nov 10 16:27:21 witherspoon openpower-update-manager[1142]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Nov 10 16:27:21 witherspoon openpower-update-manager[1142]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Nov 10 16:27:21 witherspoon openpower-update-manager[1142]: The operation failed internally.
Nov 10 16:27:21 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/control/host0/boot/one_time.
Nov 10 16:27:21 witherspoon systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Nov 10 16:27:21 witherspoon systemd[1]: Started Process Core Dump (PID 1181/UID 0).
Nov 10 16:27:22 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=dumped, status=6/ABRT
```

This error corresponds to doing a mapper query to the chassis object to see if the host is powered on. Talking with Andrew it appears that it might be due to the chassis object not being started yet, so need to add a dependency to the chassis service on the pnor updater service.

rfrandse, 2017-11-11:
<https://gerrit.openbmc-project.xyz/7883> Add dependency to the chassis service
Resolves  #2591 openpower-update-manager  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally. - OP910 1742E

amboar, 2017-11-12:
If we're going to bridge bugs, *please* edit the summary and description to remove any references to internal version descriptions, and replace them with relevant, publicly visible tags. Tagging this as ibm-internal until the issue is edited.

anoo1, 2017-11-14:
The issue was with this commit that was cherry-picked into a build: https://gerrit.openbmc-project.xyz/#/c/7122/

The commit has been refreshed to fix this issue. Closing.

closed 2017-11-14 by anoo1