openbmc/openbmc #2594: Pegas1.0 - STC:OP910:Witherspoon: yl13uf:Power Button NOT Blinking at BMC Standby

cdeadmin, 2017-11-13:


amboar, 2017-11-13:
This issue has been tagged as `ibm-internal` because it references IBM-internal tags, commits, machines or issues. This is a public issue tracker and is for the benefit of the community. If any of the details are not publicly accessible then the issue should be filed elsewhere, or reworded to reflect the relevant publicly accessible details.

amboar, 2017-11-13:
This issue also has no description. Both of these failings are not compatible with building an inclusive open source community.

rfrandse, 2017-11-17:
@vishwabmc 
Power Button was not blinking while system at BMC standby.

At BMC Standby, While the green power-on LED was blinking, the Power Button was not.
It should behaves exactly the same as the green power-on LED.
Action = [reject]
Answer = [Need More Info]

Sorry, but I am not following what is the meaning of "Power Button was not blinking".

Why would a particular button blink ?. We are blinking the PowerLED as noted by you and
that is saying that we are at STANDBY. I have not heard of anything like a button blinking.

closed 2017-11-17 by rfrandse

mdmillerii, 2017-12-11:
For future readers, the power button has an LED inside it.  The GPIO was defined as an additional indicator and added to the list of LEDs to follow power state.