openbmc/openbmc #2612: Create in Memory SDR Data Structures

zahrens, 2017-11-14:


dkodihal, 2018-01-11:
The goal of this story is to get "GetSDR" and "SDR sensor list" working with host-based ipmi sensors.

- [ ] Getting Entity ID (which is in MRW) and Entity instance (which isn't in MRW - need to code this up) in the final Witherspoon sensors yaml.
- [ ] Review each field in type 1 SDR, if we think the field is required, then code it up (pull it from MRW/hand-code in input yaml/hard-code with default values in ipmi code)

rfrandse, 2018-01-25:
<https://gerrit.openbmc-project.xyz/8604> sensor: Modify config for sensor to indicate sensor naming pattern
Resolves  #2612 Host-IPMI sensors : SDR data structures created in memory

closed 2018-02-06 by geissonator