openbmc/openbmc #2625: IPMI DCMI : Get power reading

dkodihal, 2017-11-14:


dkodihal, 2017-11-14:
- could be an aggregate of sensors
- sensor dbus name depends on hwmon config, so can't be hard coded.

dkodihal, 2018-01-11:
- [ ] Need to investigate from which D-Bus object(s)/interface(s) current power reading can be read. May need to encode this in a data file, based on design discussion.
- [ ] Populate current power reading, in Watts, in the response. Fill up other fields of the response.
- [ ] For now, set min/max/avg reading to the same value as current. Open an issue to rework this when telemetry is in.

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8573> DCMI: Populate response buffer with power reading
Resolves  #2625 IPMI DCMI : Get power reading

closed 2018-02-06 by geissonator