openbmc/openbmc #2627: GUI: OpenBMC GUI: Not able to login using Safari as browser

sivassrr, 2017-11-14:
Problem Description:
Using bluemix app trying to launch openbmc GUI using Safari as browser. 
But failed to login. Where as Chrome browser able to open the same. 

Have shared the details with Karl Vochatzer on the same. 
[image]

sivassrr, 2017-11-16:
In today's recreate, Observe that mail screen comes and goes back to login screen as a flash

iftekharuli, 2017-11-16:
@sivassrr this is not the latest version. In the latest version login changed. We have host input now 

[image]

sivassrr, 2017-11-16:
Point is Safari browser not helping to log into the main screen. Its login and logsout immediately. Probably we can try that tomorrow.

sivassrr, 2017-11-17:
Thank you @iftekharuli - With the code what you had provided able to log into BMC using safari browser, but it takes time and not coming out the below screen. 

[image]

iftekharuli, 2018-03-13:
I was able to login and perform actions using safari. Let me know if this is still and issue 

[image]

sivassrr, 2018-04-20:
Today also tried and able to use it successfully.. we can close

closed 2018-04-20 by sivassrr