openbmc/openbmc #2629: Witherspoon multi-rotor failure/ fan speed control

cdeadmin, 2017-11-14:


rfrandse, 2017-11-15:
@msbarth 
system shuts down only if there are two different fan assemblies with a failure. The system should shut down for any two combined rotor failures. (this should include two rotors within the same fan)
Matt Barth's description: each fan rotor should have a functional state to base table off of the fan rotor and not the fan.
...
Investigating the 2 implementation options that would allow each rotor within a fan to have an associated functional state. From a fan control perspective, there is minimal changes needed to support either, however how each implementation filters down affecting other areas(error logging, leds, etc..) is what needs to be looked into.

msbarth, 2017-11-16:
The implementation direction for this change will be to add child functional states for each rotor within a fan enclosure while keeping the functional state of the fan enclosure where it is now. The fan monitor application, where the functional state of a fan is managed, will add these additional child rotor states and manage the overall functional state of the fan enclosure. This will allow other applications dependent on the fan functional state to use the overall fan enclosure functional state and/or the individual rotor functional state.

- [ ] Fan monitor add functional properties for each rotor it monitors
- [ ] Init each rotor as functional at each poweron
- [ ] Fan monitor manages functional state of enclosure based on rotor(s)
(Enclosure functional when all rotors are functional, any rotor non-functional makes enclosure non-functional)

rfrandse, 2017-11-21:
<https://gerrit.openbmc-project.xyz/8004> Set tach sensor inventory functional state
Resolves  #2629 Multi-rotor failure/ fan speed control

closed 2017-12-05 by geissonator