openbmc/openbmc #2630: [xCAT] After power on or off command is sent to BMC, next command takes 15 seconds for response

cdeadmin, 2017-11-14:


msbarth, 2017-11-14:
The BMC is busy processing a large amount of sensors for fan control (ie. if OCC is Active).
Here is a trace:
```Nov 07 19:34:03 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Nov 07 19:34:06 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Nov 07 19:34:07 witherspoon phosphor-fan-control[2771]: Failed to invoke DBus method.
Nov 07 19:34:07 witherspoon phosphor-fan-control[2771]: The operation failed internally.
Nov 07 19:34:07 witherspoon phosphor-fan-control[2771]: Unable to find property
Nov 07 19:34:07 witherspoon phosphor-fan-control[2771]: Failed to invoke DBus method.
Nov 07 19:34:07 witherspoon phosphor-fan-control[2771]: The operation failed internally.
--
--
Nov 07 19:34:22 witherspoon phosphor-fan-control[2771]: The operation failed internally.
Nov 07 19:34:22 witherspoon phosphor-fan-control[2771]: Unable to find property
Nov 07 19:34:22 witherspoon phosphor-fan-control[2771]: Failed to invoke DBus method.
Nov 07 19:34:22 witherspoon phosphor-fan-control[2771]: The operation failed internally.
Nov 07 19:34:22 witherspoon phosphor-fan-control[2771]: Unable to find property
Nov 07 19:34:38 witherspoon ipmid[1259]: WATCHDOG RESET
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG RESET
Nov 07 19:34:41 witherspoon ipmid[1259]: IPMI SET_SENSOR [0x03]
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG RESET
Nov 07 19:34:41 witherspoon ipmid[1259]: IPMI GET_SENSOR_READING [0xd7]
Nov 07 19:34:41 witherspoon ipmid[1259]: IPMI GET_SENSOR_READING [0xd8]
Nov 07 19:34:41 witherspoon ipmid[1259]: Contents of a 0xc8 is 0x0a
Nov 07 19:34:41 witherspoon ipmid[1259]: IPMI SET_SENSOR [0x08]
Nov 07 19:34:41 witherspoon ipmid[1259]: IPMI SET_SENSOR [0x09]
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG RESET
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG SET Timer:[0x12C] 100ms intervals
Nov 07 19:34:41 witherspoon ipmid[1259]: Handling Netfn:[0x6], Cmd:[0x36]
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG SET Timer:[0x258] 100ms intervals
Nov 07 19:34:41 witherspoon ipmid[1259]: WATCHDOG RESET
```
(the extraneous fan control entries for properties not found is being handled under #2628)

msbarth, 2017-11-14:
The generated code for the fan control application should attempt to use another method in retrieving sensors where they may all be included with a single dbus call (ie. using getsubtree...) instead of each sensor being looked up individually. That should reduce the amount of time it takes for fan control to fully populate its needed sensor data.

msbarth, 2017-12-11:
Debugging and testing results from #2708 has taken a few days along with resulting in an additional fix required as a top priority over working this issue. I'll return to working this issue soon as I complete a fix for fan oscillations occurring in the control algo.

rfrandse, 2018-02-24:
<https://gerrit.openbmc-project.xyz/8567> Minimize service name mapper lookups
Resolves  #2630 After power on or off command is sent to BMC, next command takes 15 seconds for response

closed 2018-03-14 by geissonator