openbmc/openbmc #2631: occ-hwmon occ2-dev0: occ bus op failed rc:-5

shenki, 2017-11-15:
4.10.17-6b6e0a3593d3efba15eda8d2d824d63daf30fed4 and v1.99.10-257-ga88f18d

My machine was powered off. I rebooted the BMC, did a 'obmcutil poweron' and as the host came up the kernel hit errors:
```
sbefifo 01:01:00:06: Fatal bus access failure: -5
occ-hwmon occ2-dev0: occ bus op failed rc:-5
sbefifo 00:00:00:06: Fatal bus access failure: -5
occ-hwmon occ1-dev0: occ bus op failed rc:-5
occ-hwmon occ2-dev0: occ bus op failed rc:-5

```

Full low below.

```
Nov 15 05:04:00 ozrom1 phosphor-host-state-manager[1046]: Host State transaction request
Nov 15 05:04:00 ozrom1 systemd[1]: Starting Assert power_on LED...
Nov 15 05:04:00 ozrom1 systemd[1]: Starting Wait for Power0 to turn on...
Nov 15 05:04:00 ozrom1 systemd[1]: Started Phosphor GPIO checkstop monitor.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Power0 Host Off.
Nov 15 05:04:00 ozrom1 systemd[1]: Started Phosphor poweron watchdog.
Nov 15 05:04:00 ozrom1 systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Power0 Off.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped Assert power_on LED.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Stop Host0.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Chassis0 (Power Off).
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped Stop Power0.
Nov 15 05:04:00 ozrom1 systemd[1]: Starting Reload mboxd during power on...
Nov 15 05:04:00 ozrom1 systemd[1]: Starting Trigger CPLD to give pgood...
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped Reset CPLD to wait after VDD is powered up.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped Wait for Power0 to turn off.
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Power0 Off (Pre).
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Host0 (Stopped).
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Host0 (Stopping).
Nov 15 05:04:00 ozrom1 systemd[1]: Stopped target Stop Host0 (Pre).
Nov 15 05:04:01 ozrom1 phosphor-host-state-manager[1046]: Received signal that host is off
Nov 15 05:04:01 ozrom1 phosphor-host-state-manager[1046]: Change to Host State
Nov 15 05:04:01 ozrom1 phosphor-chassis-state-manager[996]: Received signal that power OFF is complete
Nov 15 05:04:01 ozrom1 phosphor-chassis-state-manager[996]: Change to Chassis Power State
Nov 15 05:04:01 ozrom1 systemd[1]: Started Reload mboxd during power on.
Nov 15 05:04:01 ozrom1 mboxctl[3584]: Reset: Success
Nov 15 05:04:01 ozrom1 systemd[1]: Reached target Power0 On (Pre).
Nov 15 05:04:01 ozrom1 systemd[1]: Starting Start Power0...
Nov 15 05:04:02 ozrom1 power_control.exe[995]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Nov 15 05:04:02 ozrom1 power_control.exe[995]: PowerControl: setting power up BMC_POWER_UP to 1
Nov 15 05:04:02 ozrom1 systemd[1]: Started Start Power0.
Nov 15 05:04:02 ozrom1 system_manager.py[924]: Running System State: HOST_POWERING_ON
Nov 15 05:04:02 ozrom1 systemd[1]: Starting Perform AVS bus workaround on VRMs...
Nov 15 05:04:02 ozrom1 systemd[1]: Started Perform AVS bus workaround on VRMs.
Nov 15 05:04:02 ozrom1 systemd[1]: Starting Disable the AVS bus on the VRMs...
Nov 15 05:04:03 ozrom1 systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Nov 15 05:04:03 ozrom1 systemd[1]: Started Trigger CPLD to give pgood.
Nov 15 05:04:03 ozrom1 systemd[1]: Started Disable the AVS bus on the VRMs.
Nov 15 05:04:03 ozrom1 systemd[1]: Starting Apply voltage overrides to VRMs...
Nov 15 05:04:03 ozrom1 vrm.sh[3623]: rail        set        read        current
Nov 15 05:04:03 ozrom1 vrm.sh[3623]: -------        ------- ------- -------
Nov 15 05:04:03 ozrom1 power_control.exe[995]: PowerControl: pgood: 1, setting reset BMC_CP0_RESET_N to 1
Nov 15 05:04:03 ozrom1 power_control.exe[995]: PowerControl: pgood: 1, setting reset BMC_CP0_PERST_ENABLE_R to 1
Nov 15 05:04:03 ozrom1 system_manager.py[924]: Running System State: HOST_POWERED_ON
Nov 15 05:04:03 ozrom1 systemd[1]: Started Wait for Power0 to turn on.
Nov 15 05:04:04 ozrom1 systemd[1]: Started Assert power_on LED.
Nov 15 05:04:04 ozrom1 systemd[1]: Reached target Power0 On.
Nov 15 05:04:04 ozrom1 systemd[1]: Reached target Power0 (On).
Nov 15 05:04:04 ozrom1 systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Nov 15 05:04:04 ozrom1 systemd[1]: Started Phosphor Fan Monitor Initialization.
Nov 15 05:04:04 ozrom1 systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Nov 15 05:04:04 ozrom1 systemd[1]: Started Phosphor Fan Control Initialization.
Nov 15 05:04:04 ozrom1 systemd[1]: Starting Scan FSI devices...
Nov 15 05:04:04 ozrom1 systemd[1]: Started Phosphor Fan Control Daemon.
Nov 15 05:04:06 ozrom1 kernel: occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[... 371 lines cut ...]
Nov 15 05:05:44 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:44 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:44 ozrom1 kernel[860]: [15961.520000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:44 ozrom1 kernel[860]: [15961.530000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:46 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:46 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:46 ozrom1 kernel[860]: [15962.670000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:46 ozrom1 kernel[860]: [15962.690000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:47 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:47 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:47 ozrom1 kernel[860]: [15963.790000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:47 ozrom1 kernel[860]: [15963.820000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:48 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:48 ozrom1 kernel[860]: [15964.910000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:48 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:48 ozrom1 kernel[860]: [15964.950000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:49 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:49 ozrom1 kernel[860]: [15966.030000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:49 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:49 ozrom1 kernel[860]: [15966.070000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:50 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:50 ozrom1 kernel[860]: [15967.160000] occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:50 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5
Nov 15 05:05:50 ozrom1 kernel[860]: [15967.200000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
```

eddiejames, 2017-11-15:
Nov 15 05:05:18 ozrom1 kernel: sbefifo 01:01:00:06: Fatal bus access failure: -5
Nov 15 05:05:18 ozrom1 kernel: occ-hwmon occ2-dev0: occ bus op failed rc:-5
Nov 15 05:05:20 ozrom1 kernel: sbefifo 00:00:00:06: Fatal bus access failure: -5
Nov 15 05:05:20 ozrom1 kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-5

Looks like FSI failures on to both SBEFIFOs. OCC accesses keep failing because SBEFIFO cannot recover; the FIFO state is unknown after a bus access failure. Hwmon poller should stop after a couple of errors here.

Why did FSI fail? @cbostic has been looking into occasional bus failures, probably related.

mine260309, 2017-11-20:
I can reproduce this issue on PASS2 Romulus, but the `occ1` or `occ2` are random.

Steps:
1. AC cycle, and power on, it's working fine;
2. Power off host;
3. Power on host.

After host is running and  Phosphor Hwmon is running on occ, I get below logs
```
[  621.550000] sbefifo 00:00:00:06: Fatal bus access failure: -5
[  621.550000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
[  625.600000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
[  630.810000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
[  632.460000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
...
```
Sometimes it's `occ2-dev0`, sometime both occ1 and occ2 are getting the log.

mine260309, 2017-11-20:
It can be manually recovered by following commands:
```
echo occ1-dev0 > /sys/bus/platform/drivers/occ-hwmon/unbind
# If we then do bind, it fails with occ-hwmon occ1-dev0: occ bus op failed rc:-5
openpower-proc-control scanFSI
echo occ1-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind
echo occ2-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind
```

mine260309, 2017-11-24:
According to @eddiejames:
> It's an fsi failure. once sbefifo hits it once, its done until fsi rescan.

So probably `openpower-occ-control` need related change to do fsi rescan?

cbostic, 2018-01-30:
This appears to be a duplicate of https://github.com/openbmc/openbmc/issues/2343. 

The FSI bus errors were due to a small window during power on where some form of bus contention was going on between the BMC and Hostboot.   Once out of this time window the fails stopped and never happened again.

The ClearQuest defect bridged from issue 2343 was sent on to HostBoot to debug what may be touching the FSI bus during this period.

mine260309, 2018-03-30:
With latest OpenBMC (kernel on dev-4.13), the issue is gone.
The occ/fsi driver are updated on dev-4.13 and it seems to have fixed this issue.

amboar, 2018-04-18:
Closing this based on @mine260309's feedback.

closed 2018-04-18 by amboar