openbmc/openbmc #2633: SBE side switch failure

kantvijay, 2017-11-15:
Tests.Test Sbe Side Switch.Test SBE Side Switch
Documentation: Trigger watchdog errors on the host until it side switches.

INFO | Argument types are: <type 'int'> <type 'unicode'> 

FAIL : 3 != 2

kantvijay, 2017-11-15:
http://tulpok01p4.aus.stglabs.ibm.com:8080/jenkins/job/OpenBMC/job/automated-test/job/helper-jobs/job/CT_Run/270/robot/report/log.html#s1-s22-t1

kantvijay, 2017-11-15:
PNOR firmware level:         driver content      

    
display pnor FW level using this cmd:       "cat /var/lib/phosphor-software-manager/pnor/ro/VERSION" 
  
    IBM-witherspoon-ibm-OP9_v1.19_1.73
    op-build-v1.19-224-g1fe4582
    buildroot-2017.08-8-g5e23247
    skiboot-v5.9
    hostboot-5090f7e
    linux-4.13.8-openpower2-pc69996b
    petitboot-v1.6.2-pdfbb2d0
    machine-xml-755040c
    occ-dbb4d7e
    hostboot-binaries-013fbba
    capp-ucode-p9-dd2-v2
    sbe-e2cfab8 
                 

BMC firmware level :                     driver content           

        display BMC FW level via ssh session on the BMC , using this cmd    root@witherspoon:~# cat /etc/os-release 

    id:                   openbmc-phosphor
    name:             Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
    version:          ibm-v1.99.10
    version_id:     ibm-v1.99.10-113-g65edf7d-r10-0-gcdf7635
    pretty_name: Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10
    build_id:         ibm-v1.99.10-113-g65edf7d-r10

gkeishin, 2017-11-15:
Please provide the journald log for this and we want to verify first in automation if the failure is because the watchdog inject error didnt create one or there is something else,..

closed 2017-11-15 by gkeishin