openbmc/openbmc #2636: OP910: systemd output showing warning and errors

cdeadmin, 2017-11-15:


geissonator, 2017-11-15:
We have some low-priority clean up still to do within our systemd service/target files.  
This is running our v2.0 tagged code.  The .mount files are created dynamically, we may want a separate issue to solve that one.  The other warnings are just target and service files that need to be cleaned up.   A quick look didn't show any major issues (mostly just legacy targets)
```
[witherspoon]# systemctl status --all | cat | grep -e found
             \u2502   \u2514\u25003312 grep -e found
Warning: etc-machine\x2did.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-pnor\x2dprsv.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-pnor\x2dro\x2d7a9ef57c.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-pnor\x2dro\x2d8764baa3.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-pnor\x2drw\x2d7a9ef57c.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-pnor\x2drw\x2d8764baa3.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-rofs\x2d8e0b4dba.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: media-rofs\x2daa2733ca.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: usr-local-share-pnor.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Unit dropbear.service could not be found.
           \u2514\u25003312 grep -e found
Unit fsi-hub-scan@0.service could not be found.
Unit plymouth-quit-wait.service could not be found.
Unit plymouth-start.service could not be found.
             \u2502   \u2514\u25003312 grep -e found
             \u2514\u25003312 grep -e found
           \u2502   \u2514\u25003312 grep -e found
Unit cryptsetup.target could not be found.
Unit obmc-chassis-poweroff0.target could not be found.
Unit obmc-chassis-reset@0.target could not be found.
Unit obmc-chassis-reset@i.target could not be found.
Unit obmc-chassis-reset@obmc-flash-bios-updatesymlinks.target could not be found.
Unit obmc-host-stop0.target could not be found.
Unit obmc-stop-host@0.target could not be found.
Unit obmc-stop-host@i.target could not be found.
Unit syslog.target could not be found.
```

spinler, 2018-06-29:
I'll fixup some of these.

spinler, 2018-06-29:
These commits:
https://gerrit.openbmc-project.xyz/#/c/11354/
https://gerrit.openbmc-project.xyz/#/c/11355/
https://gerrit.openbmc-project.xyz/#/c/11356/

fix the following:
```

Unit obmc-flash-bios-check-clearvolatile.service could not be found.
Unit obmc-chassis-poweroff0.target could not be found.
Unit obmc-chassis-reset@0.target could not be found.
Unit obmc-host-stop0.target could not be found.
Unit obmc-stop-host@0.target could not be found.
Unit fsi-hub-scan@0.service could not be found.
```

closed 2018-07-18 by rfrandse