openbmc/openbmc #2641: phosphor-networkd does not populate `domain` option in /etc/resolv.conf

shenki, 2017-11-16:
When booting the BMC in an environment where network configuration comes through DHCP, /etc/resolf.conf is populated with htis:
```
# cat /etc/resolv.conf 
### Generated by phosphor-networkd ###
nameserver 10.61.2.7 10.61.2.1
```
If I run udhcpc, I get those entries plus the `domain` option:
```
# cat /etc/resolv.conf 
domain ozlabs.ibm.com
nameserver 10.61.2.7
nameserver 10.61.2.1
```
The domain option is expected in the networking environment this machine is placed in.

We need to ensure that phosphor-networkd is obeying all of the configuration information passed to it by the dhcp server. Ideally we would be re-using existing userspace components for this.

ratagupt, 2017-12-12:
This is not a bug.we have not enabled configuring the local domain name through user space application as we don't need this till now..

Configuring domain in /etc/resolv.conf only gets you that this domain will be looked first for any resolver query..
We are obeying all the configuration that dhcpd is sending but we don't have user space configuration for domain name as we don't see this requirement.

shenki, 2017-12-12:
I have opened this ticket as it is a requirement for the environment I use OpenBMC in.

Do you know what change(s) need to be made in order to make this work?

bradbishop, 2017-12-19:
The problem here is that we are not running systemd-resolved or something like it (like udhcpc).  It seems like enabling that should be really easy.  I think the second part to that would be removing the code from phosphor-networkd that is emitting the file.

I agree with Joel, we should be letting one of these other applications manage the content of /etc/resolve.conf.  phosphor-networkd was only supposed to be providing a dbus API for configuration and status of all this stuff...it shouldn't be sticking its fingers in any deeper than that.

shenki, 2018-10-11:
It looks like openbmc has started using systemd-resolve to manage DNS. This issue still exists with the current openbmc master.

shenki, 2019-04-29:
Can we get this added to someone's task list @bradbishop ?

closed 2020-06-12 by stale[bot]