openbmc/openbmc #2643: witherspoon: power button's integreated LED should blink at standby on power 9 just like it did on power 8 systems.

cdeadmin, 2017-11-16:


vishwabmc, 2017-11-23:
```This LED wired to GPIO R5 is integrated with power button

Change-Id: I899405013fbb6b93783dec59090e86453a5ec383

---
 arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts | 4 ++++
 1 file changed, 4 insertions(+)

diff --git a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts b/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
index 68058ec..d25c7dc 100644
--- a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
+++ b/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
@@ -143,6 +143,10 @@
 		rear-power {
 			gpios = <&gpio ASPEED_GPIO(N, 3) GPIO_ACTIVE_LOW>;
 		};
+
+		power-button {
+			gpios = <&gpio ASPEED_GPIO(R, 5) GPIO_ACTIVE_LOW>;
+		};
 	};

 	fsi: gpio-fsi {```

closed 2018-03-01 by rfrandse

zhuifeng091, 2018-11-29:
if boot up system from openBMC, gpio first pull low, sleep 1s then pull down, how to configure?