openbmc/openbmc #2644: OpenBMC:Witherspoon:FVT- Seen 500 internal error and network configuration not getting popultaed properly

cdeadmin, 2017-11-16:


rfrandse, 2017-11-16:
<https://gerrit.openbmc-project.xyz/7931> Increase the waiting time for refreshing objects
Resolves  #2644 OpenBMC:Witherspoon:FVT- Seen 500 internal error and network configuration not getting popultaed properly

rfrandse, 2017-11-16:
@ratagupt 
Here the refreshing complete at 09:39:27
Nov 16 09:39:27 witherspoon phosphor-network-manager[2783]: Refreshing complete.
and the next dbus request comes at the same second..

Nov 16 09:39:27 phosphor-gevent[1158]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 858, in _handle

seems mapper has not been taken care the newly added dbus object.

Increase the waiting time for refreshing objects Increase the waiting time for refreshing the objects as event for link local object comes late by sec or two.

closed 2017-11-28 by geissonator