openbmc/openbmc #2645: MFG:Witherspoon, ReadFailureError ..sbefifo1-dev0/occ1-dev0, 1742C

cdeadmin, 2017-11-16:


spinler, 2017-11-16:
Several systems are seeing an ENODATA returned by the P9 OCC device driver in phosphor-hwmon-readd.  This errno isn't on the retry list, so phosphor-hwmon will log an error, restart, and then things seem to be fine.

It is suspected that the host is doing something with the P9 SBE FIFO at the time of  the failure, so I'm going to add this errno to the retry list so hopefully the retries can get a clean read.

rfrandse, 2017-11-16:
<https://gerrit.openbmc-project.xyz/7952> Retry on ENODATA errors
Resolves  #2645 ENODATA ReadFailureError error on OCC ..sbefifo1-dev0/occ1-dev0

amboar, 2017-11-20:
Are we trying to root-cause the problem? The suggested patch is a work-around.

spinler, 2017-11-20:
@amboar We ran out of time, not that putting this in precludes anyone from still investigating.

amboar, 2017-11-21:
@spinler yep - are you happy to leave this open?

closed 2017-12-12 by geissonator