openbmc/openbmc #2646: Deletion of an error log entry results in xyz.openbmc_project.Logging.service: Failed with result 'core-dump'

cdeadmin, 2017-11-17:


dkodihal, 2017-11-17:
This helps understand the problem :

```
    "/xyz/openbmc_project/logging/entry/173": {
      "AdditionalData": [],
      "Id": 0,
      "Message": "", 
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency",
      "Timestamp": 0,
      "associations": []
    },
    "/xyz/openbmc_project/logging/internal/manager": {}
```

The error object has a valid path, but properties with default values. A delete on this would try to retrieve the corresponding error log persisted file (we use Cereal for persistence) by the id property, but that has a value of 0, which is unexpected.

a) Since we can't recreate this at will, my line of attack is to figure what can cause the error id to be 0 in the d-bus object. The error id is controlled by the log-manager (and not callers of the error API), and it's an incrementing number starting with 1. I'm ruling out corruption on D-Bus bus itself, because that result in something more pervasive.

b) So the next possibility is when an error log was created on the d-bus, and then persisted, it was for some reason persisted with values not matching d-bus properties. This could be a Cereal problem, or a filesystem problem. I'm leaning towards the former, because I see default-ctor provided values for the error log properties in the persisted file, not some junk.

c) If b) happens, then during the restore of such an error, we will get into what we see today, that is a proper d-bus error log path, but with default property values. The best way to handle this IMO would be to implement a restoreSanity() check, which you would call right after deserialize() succeeds, and in the new function, check if the error id() property matches the error object d-bus path name (In the case of this defect, you would see 0 != 173), and in which case, ignore the error and log a trace saying the persisted error is not reliable. restoreSanity() could just be a lambda within restore().

d) Since this could be a Cereal problem, we should probably apart from the fix above, run a stress test to create errors in a loop and verify that the persisted file always matches. If we hit fails, a bug could be submitted upstream to Cereal.

rfrandse, 2017-11-17:
<https://gerrit.openbmc-project.xyz/7967> Perform sanity checking while restoring error entries
Resolves  #2646 Deletion of an error log entry results in xyz.openbmc_project.Logging.service: Failed with result 'core-dump'

closed 2017-11-20 by geissonator