openbmc/openbmc #2648: OpenBMC GUI: Full hostname not displayed correctly in the BMC IP address

sivassrr, 2017-11-17:
@iftekharuli With full hostname not displayed correctly in BMC IP address
[image]

sivassrr, 2017-11-26:
May be fix can as follows under the app/common/dataService.js
```
            this.getServerId = function(){
              /*  return this.host.replace(/[^/D]+/m,"");  */
                 return this.host.replace(/^https?\:\/\//ig,"");
            }
```
It works.

sivassrr, 2017-11-27:
[image]

sivassrr, 2017-11-27:
[image]

rfrandse, 2018-01-27:
<https://gerrit.openbmc-project.xyz/8693> Fix to get full BMC IP address / hostname
Resolves  #2648 OpenBMC GUI: Full hostname not displayed correctly in the BMC IP address

closed 2018-02-01 by geissonator