openbmc/openbmc #2649: on click logout -  not happening

sivassrr, 2017-11-17:
Continuation of  previous issue https://github.com/openbmc/openbmc/issues/2648

On click logout - not happening, Had to refresh and then it logs off. 

Expectation is to logoff after clicking logout button. ( Refresh should be avoided).

iftekharuli, 2018-03-13:
This has been fixed. Let me know

amboar, 2018-03-14:
@iftekharuli Please could link to the patch fixing the issue when making these comments? That way others can figure out what they need to do to get the fix.

iftekharuli, 2018-03-21:
Duplicate https://github.com/openbmc/openbmc/issues/2737 
Also link for the the fix https://gerrit.openbmc-project.xyz/#/c/8828/

gtmills, 2018-04-19:
This was a duplicate of #2737 and has been fixed with https://gerrit.openbmc-project.xyz/#/c/8828/.

closed 2018-04-19 by gtmills