openbmc/openbmc #2653: Witherspoon:FVT:910- BMC went into bad state, not responding to REST

cdeadmin, 2017-11-20:


rfrandse, 2017-11-20:
<https://gerrit.openbmc-project.xyz/7996> Fix BMC went into bad state, not responding to REST
Fixes  #2653 Witherspoon:FVT:910- BMC went into bad state, not responding to REST

rfrandse, 2017-11-20:
@devenrao 
System went into bad state, and REST is not responding on latest build
It happens when we try to set nameserver.
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  PUT  -d '{"data": "IP_ADDRESS"}' https://wsbmc007/xyz/openbmc_project/network/eth0/attr/Nameservers
...
this is only happening when user tries to give the bad data.
Rest server goes into bad state and stops responding.Can not get much data as journal logs were not coming after restarting journald.
...
I tried with the correct format then it works fine and the network dbus object gets updated.
However the underlying functionality will not work as static DNS code is not there in the build.

This defect is for why the rest server stops responding.

Correct format to set the DNS
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  PUT  -d '{"data": ["IP_ADDRESS1","x.x.x.x."]}' https://wsbmc007/xyz/openbmc_project/network/eth0/attr/Nameservers
Seems need to look at the bad path what is happening with the rest server..But on top of that we need some fix/workaround which enables the journald log on the bmc.

amboar, 2017-11-20:
@rfrandse @devenrao It would be handy if you could include the earliest known affected version in the issue as well as the issue description (referencing a change in openbmc/openbmc master)

closed 2017-12-12 by geissonator