openbmc/openbmc #2654: SUPPORT FOR AST1250

radhikapradeep, 2017-11-20:
Hi ,

We would like to know if the IBM openbmc code support ast1250 ?

In the directory "https://github.com/openbmc/openbmc/tree/master/meta-openbmc-bsp/meta-aspeed" , we found only ast2400 and ast2500. 

Can ast1250 support also be included in the code ?

Thanks and Regards,
Radhika

gtmills, 2018-08-20:
OpenBMC currently has no support for ast1250 and I have not heard of any plans to support it.

mdmillerii, 2018-09-17:
While no one has tried an ast1250, it should be a subset of the ast2400 without VGA or PCI-e.

In theory it should be similar to enabling the ast2520 from the ast2500, which would involve removing the nodes from the device tree and making sure neither the hard coded resets nor the u-boot code try to enable the disabled blocks.

That said, I am not aware of anyone who has a platform to test.

gtmills, 2018-10-12:
Closing since not a defect or a request for a feature. In the future, this is a good question for IRC or the mailing list, as it's not a bug or feature request. You can sign yourself up for the mailing list at https://lists.ozlabs.org/listinfo/openbmc 

Thanks!!

closed 2018-10-12 by gtmills